ccMSC/ckb #414: Daemon dies during sleep

Paul92, 2016-07-12:
Quite often I leave my laptop in sleep mode. However, once I restore it, the daemon is stopped and I need to restart it (this does not happen if I shut down the computer).

Apparently, the problem is that the daemon cannot find the device and is killed with a segfault. Below is the output from systemctl status. I am using K70 RGB with archlinux.

[image]

light2yellow, 2016-07-18:
Hi. Put this in your `ckb-daemon.service` and execute `systemctl daemon-reload`:
```
[Unit]
Description=Corsair Keyboard Daemon
After=suspend.target

[Service]
ExecStart=/usr/bin/ckb-daemon
Restart=always
StandardOutput=syslog
StandardError=inherit
RemainAfterExit=yes

[Install]
WantedBy=multi-user.target sleep.target
```
Tell if it works.

Paul92, 2016-07-20:
Thanks for your reply. Unfortunately, it still doesn't work (the daemon is still killed with a SEGV, but there are no diagnostic messages as before).

tatokis, 2016-07-20:
What version of systemd are you on, and what USB controller are you 
using?
I am on systemd 229 and resume works fine using an Etron EJ168 USB3 
controller, so I can't reproduce it.

I suspect it is a timing issue.

Paul92, 2016-07-20:
Systemd version is 330.

The USB controller is a Intel corporation 7 series/C210 series chipset familiy (this is what lspci reports, not sure it's the best way to find the controller). The keyboard is connected on USB 3.0, using just the keyboard connector.

tatokis, 2016-07-20:
I just tried my K70 RGB on laptop with an Intel 8 series/C220 controller, both on USB2 and 3, and had no segfaults.
Kill the daemon, then run it through gdb and then see if you can get a backtrace after resuming.
```
sudo killall -9 ckb-daemon && sudo gdb ckb-daemon
run
```
Make sure you have another keyboard in order to type 'run' above, as killing the daemon prevents input from the corsair keyboard.
Then, once it segfaults
```
bt full
```

light2yellow, 2016-07-20:
@tatokis shouldn't ckb be fully recompiled with `-g` previously?

Paul92, 2016-08-16:
Ok, so since no one clarified the debugging thing, I am adding some debug output from systemctl that includes the stack trace:

`Aug 16 19:09:47 localhost kernel: usbhid 3-1:1.2: couldn't find an input interrupt endpoint
Aug 16 19:09:47 localhost kernel: hid-generic 0003:1B1C:1B2F.0002: usb_submit_urb(ctrl) failed: -1
Aug 16 19:09:47 localhost kernel: hid-generic 0003:1B1C:1B13.0004: usb_submit_urb(ctrl) failed: -1
Aug 16 19:09:47 localhost kernel: usbhid 3-2:1.2: couldn't find an input interrupt endpoint
Aug 16 22:18:38 pc-81.home cinnamon-screensaver-dialog[10764]: gkr-pam: the password for the login keyring was invalid.
Aug 16 22:57:42 localhost.localdomain kernel: [drm:intel_set_cpu_fifo_underrun_reporting [i915]] *ERROR* uncleared fifo underrun on pipe B
Aug 16 22:57:43 localhost.localdomain kernel: usbhid 3-2:1.2: couldn't find an input interrupt endpoint
Aug 16 22:57:44 localhost.localdomain systemd-coredump[16764]: Process 435 (ckb-daemon) of user 0 dumped core.

                                                               Stack trace of thread 474:
                                                               #0  0x0000000000404e75 _ledthread (ckb-daemon)
                                                               #1  0x00007f903aee2454 start_thread (libpthread.so.0)
                                                               #2  0x00007f903a3827df __clone (libc.so.6)

                                                               Stack trace of thread 435:
                                                               #0  0x00007f903b2ca4b6 udev_monitor_receive_device (/usr/lib/libudev.so.1.6.5)
                                                               #1  0x000000000040a232 usbmain (ckb-daemon)
                                                               #2  0x00000000004070d4 main (ckb-daemon)
                                                               #3  0x00007f903a2ba291 __libc_start_main (libc.so.6)
                                                               #4  0x0000000000402189 _start (ckb-daemon)
-- Subject: Process 435 (ckb-daemon) dumped core
-- Defined-By: systemd
-- Support: http://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- Documentation: man:core(5)
--
-- Process 435 (ckb-daemon) crashed and dumped core.
--
-- This usually indicates a programming error in the crashing program and
-- should be reported to its vendor as a bug.
Aug 16 22:57:53 pc-81.home kernel: hid-generic 0003:1B1C:1B13.0008: usb_submit_urb(ctrl) failed: -1
Aug 16 22:57:53 pc-81.home kernel: usbhid 3-2:1.2: couldn't find an input interrupt endpoint
`

Not sure whether this is related to my setup or not, but it seems that the keyboard doesn't respond anymore (I also have a sabre mouse, which I believe is the source of the second interrupt endpoint exception).

Since I have a programming background, I will be more than happy to try to trace this myself if anyone provides some guidance.