ccMSC/ckb #415: m65 rgb pro not detected

TheChellgames, 2016-07-16:
I realize that its not under the supported devices, but I am wondering if it will be soon. I can imagine its not an easy process but i would greatly appreciate it. 

Thank you for this amazing software!

tatokis, 2016-07-16:
There is experimental support for the M65 RGB Pro on the newdev branch.

TheChellgames, 2016-07-17:
@tatokis How do I go about accessing that?

TheChellgames, 2016-07-18:
@tatokis how do i find that?

lukeberry99, 2016-07-20:
@TheChellgames there are currently no downloadable binaries, you'll have to compile it yourself.

TheChellgames, 2016-07-21:
@lukeberry99 Can you or someone walk me through or help me do that please

andrew904, 2016-10-16:
@lukeberry99 Having the same problems as @TheChellgames and need a walkthrough, I'm quite new.

lmansur, 2016-10-20:
## This is how i did in Ubuntu:

### Uninstall previous ckb version:
For Ubuntu 15.04 and above
```
sudo systemctl stop ckb-daemon
sudo rm -f /usr/lib/systemd/system/ckb-daemon.service
```
For Ubuntu earlier than 15.04
```
sudo service ckb-daemon stop
sudo rm -f /etc/init/ckb-daemon.conf
```

Now do this regardless of your Ubuntu version
```
sudo rm -f /usr/bin/ckb /usr/bin/ckb-daemon /usr/share/applications/ckb.desktop /usr/share/icons/hicolor/512x512/apps/ckb.png
sudo rm -rf /usr/bin/ckb-animations
```
Your old ckb is uninstalled.
### Installing new version
Download/clone [this branch](https://github.com/ccMSC/ckb/tree/newdev) and extract it somewhere.
Move to the extracted location and double click _quickinstall_ file ( or do the equivalent in terminal ). This will install the new branch and M65 Pro should work.

The process should be similar in OSX, just follow the documentation.

schneems, 2016-10-20:
According to #356 support for the pro model hasn't been implemented yet

schneems, 2016-10-20:
Never mind, was only looking at master, didn't check `newdev` thanks!

LxFx, 2016-11-12:
Happy user of the newdev branch for my M65 RGB Pro. When will this branch be merged with the master branch?

ags911, 2016-12-03:
Hey LxFx, is it working for you? I have the same build but it's not being detected. Do you mind sharing the zip here to try please?

LxFx, 2016-12-03:
Hi, yes everything works. I'm on Arch Linux. I just cloned the newdev branch and adapted the git PKGBUILD from the AUR and compiled successfully. I can share the PKGBUILD if that helps?

ags911, 2016-12-03:
That would be amazing bud thanks. I'm on OSX though would it work?

LxFx, 2016-12-05:
I'm quite sure it won't. But nevertheless, here goes...

https://drive.google.com/file/d/0B98pOitK_uzlSk5XQnA0cEptRWM/view?usp=sharing

ags911, 2016-12-05:
Thanks, I'll give it a try.

chrylis, 2017-01-18:
I have the Pro and am not seeing it as of the latest AUR package, which is geb1fe4e. @LxFx, do you know if the `ckb-git-latest` version includes this support?

LxFx, 2017-01-18:
No, you'll need to clone the newdev branch and use my PKGBUILD linked above and ckb-git.install from the AUR to build it. Should still work.

By the way, you can save your favorite profile/config in the mouse itself. So you don't really need this software anymore, unless you want to change the LED theme from time to time.