ccMSC/ckb #419: ckb segfault in Ubuntu 16.04

kaibacorp, 2016-07-23:
Just upgraded from 15.10 and started experiencing this error, similar to #296. Firmware version 2.04 and ckb-beta-0.2.6 with Strafe. 

Here is the backtrace:
```
(gdb) run
Starting program: /home/kai/ckb/bin/ckb 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffe8875700 (LWP 22157)]
[New Thread 0x7fffdd3c4700 (LWP 22188)]
[New Thread 0x7fffd71a1700 (LWP 22190)]
[New Thread 0x7fffd68e7700 (LWP 22191)]
[New Thread 0x7fffd59f5700 (LWP 22222)]
[New Thread 0x7fffd51f4700 (LWP 22227)]
Downloaded new firmware list. 6 entries found.
[Thread 0x7fffd51f4700 (LWP 22227) exited]
[New Thread 0x7fffd51f4700 (LWP 32512)]
[New Thread 0x7fffc5df8700 (LWP 32513)]
[New Thread 0x7fffb78da700 (LWP 32518)]
ckb: Fatal IO error 10 (No child processes) on X server :0.
QProcess: Destroyed while process ("sh") is still running.

Thread 1 "ckb" received signal SIGSEGV, Segmentation fault.
0x00007ffff5fed371 in QSocketNotifier::setEnabled(bool) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
(gdb) bt full
#0  0x00007ffff5fed371 in QSocketNotifier::setEnabled(bool) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
No symbol table info available.
#1  0x00007ffff5ee7c8b in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
No symbol table info available.
#2  0x00007ffff5f47915 in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
No symbol table info available.
#3  0x00007ffff5ee255f in QProcess::waitForFinished(int) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
No symbol table info available.
#4  0x00007ffff5ee2d7e in QProcess::~QProcess() () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
No symbol table info available.
#5  0x00007ffff4edefe8 in __run_exit_handlers (status=1, listp=0x7ffff52685f8 <__exit_funcs>, run_list_atexit=run_list_atexit@entry=true)
    at exit.c:82
        atfct = <optimized out>
        onfct = <optimized out>
        cxafct = <optimized out>
#6  0x00007ffff4edf035 in __GI_exit (status=<optimized out>) at exit.c:104
No locals.
#7  0x00007ffff41ecea1 in ?? () from /usr/lib/x86_64-linux-gnu/libgdk-x11-2.0.so.0
No symbol table info available.
#8  0x00007ffff78c2cee in _XIOError () from /usr/lib/x86_64-linux-gnu/libX11.so.6
No symbol table info available.
#9  0x00007ffff78c057d in _XEventsQueued () from /usr/lib/x86_64-linux-gnu/libX11.so.6
No symbol table info available.
#10 0x00007ffff78b1f47 in XPending () from /usr/lib/x86_64-linux-gnu/libX11.so.6
No symbol table info available.
#11 0x00007ffff41e2e2e in ?? () from /usr/lib/x86_64-linux-gnu/libgdk-x11-2.0.so.0
No symbol table info available.
#12 0x00007ffff3b3b92d in g_main_context_prepare () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
No symbol table info available.
#13 0x00007ffff3b3c2cb in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
No symbol table info available.
#14 0x00007ffff3b3c4ac in g_main_context_iteration () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
No symbol table info available.
#15 0x00007ffff6008a7f in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
No symbol table info available.
#16 0x00007ffff5fafdea in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
No symbol table info available.
---Type <return> to continue, or q <return> to quit---
#17 0x00007ffff5fb7e8c in QCoreApplication::exec() () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
No symbol table info available.
#18 0x000000000041f15c in main (argc=1, argv=0x7fffffffd128) at main.cpp:124
        a = <incomplete type>
        background = false
        w = {<QMainWindow> = {<No data fields>}, static staticMetaObject = {d = {superdata = 0x7ffff6dc8e80 <QMainWindow::staticMetaObject>, 
              stringdata = 0x527920 <qt_meta_stringdata_MainWindow>, data = 0x527c40 <qt_meta_data_MainWindow>, 
              static_metacall = 0x4ec220 <MainWindow::qt_static_metacall(QObject*, QMetaObject::Call, int, void**)>, relatedMetaObjects = 0x0, 
              extradata = 0x0}}, static mainWindow = 0x7fffffffcf70, settingsWidget = 0xa0fc80, 
          kbWidgets = {<QListSpecialMethods<KbWidget*>> = {<No data fields>}, {p = {static shared_null = {ref = {atomic = {_q_value = -1}}, 
                  alloc = 0, begin = 0, end = 0, array = {0x0}}, d = 0xe97f40}, d = 0xe97f40}}, restoreAction = 0xa058a0, closeAction = 0xa058c0, 
          unityDesktop = false, indicator = 0x7fffffffcff0, indicatorMenu = 0x764bb8, indicatorMenuQuitItem = 0x4ec207
     <__static_initialization_and_destruction_0(int, int)+67>, indicatorMenuRestoreItem = 0x7fffffffd138, trayIconMenu = 0xa05fc0, 
          trayIcon = 0xa06a00, ui = 0x9ef780}

```

tatokis, 2016-08-24:
I installed the Qt5 debug symbols, but I didn't find anything that could cause this.

However, I noticed something that may help resolve this.
I was getting constant segfaults, at random intervals but less than an hour, with the same backtrace (and sh error), only when I had a static profile set and active.

To elaborate on that, I have one profile with my usual animations, and another one with just a static colour profile, which is also saved to hardware.

Switching to the one with the animations stopped all the segfaults, switching back to the static/hardware one started it all over.

Can anyone affected, confirm this?

kaibacorp, 2016-08-25:
I set my profile to 'ckb Demo' and used the Rainbow animation, instead of my usual static colour profile. It's been running for a few hours and hasn't crashed.

jkaart, 2016-08-31:
I have this problem also!

olofson, 2016-10-01:
See also #449/#313. (Keyboard disconnects repeatedly on laptop.) Workaround for that also turns out to be having an animation playing. Coincidence...?

No disconnects on my Ubuntu Studio workstation (never had that issue in this one), but instead, I have these segfaults now. I'll report back on whether or not playing an animation avoids it here.

tatokis, 2016-10-01:
Interesting.
It's still very odd though, as I couldn't find anything wrong with 
QProcess calling up sh.

I've been running animations ever since I last replied and I have had 
no segfaults, so it is definitely related to that.

olofson, 2016-10-02:
Update: After many hours with no segfaults, I'd say having an animation playing avoids the issue here too. (Ubuntu Studio 16.04.)

olofson, 2016-10-19:
Update 2: Seems like I no longer need to have an animation running to prevent ckb segfaults after the recent *buntu upgrade of the Qt libs. I've restored my previous config (fixed colors), and it's been stable for a few hours now.