ccMSC/ckb #421: Corsair K65 RapidFire RGB; No device detected

Johnson-Z, 2016-07-23:
Corsair K65 RapidFire RGB; No device detected

Johnson-Z, 2016-07-23:
Can someone help?

kaibacorp, 2016-07-23:
Have you gone through the troubleshooting in [General > Common Issues](https://github.com/ccMSC/ckb#general) ?

Johnson-Z, 2016-07-23:
so I reboot?, but how

kaibacorp, 2016-07-23:
Reboot refers to just restarting the computer.

Reinstallation depends if you are on [Linux](https://github.com/ccMSC/ckb#uninstalling) or [OS X](https://github.com/ccMSC/ckb#uninstalling-1).

Johnson-Z, 2016-07-23:
Oh, but I tried

Johnson-Z, 2016-07-23:
My computer is mac~

Johnson-Z, 2016-07-23:
I tried everything suggested

kaibacorp, 2016-07-23:
Ah alright, you'll have to wait until someone who is more knowledgeable than me to come along to help!

Johnson-Z, 2016-07-25:
Anyone's HELP

maddrag0n, 2016-07-26:
same issue with K65 Rapidfire

mnori, 2016-08-30:
Got this problem too. Updated the firmware but still no worky.