ccMSC/ckb #422: Corsair K70 RGB not connecting with ckb application

LordPeyton, 2016-07-24:
This will be my second post on my keyboard not being able to connect to the ckb program. I don't know anything about code, so it would be great if someone could give me a step-by-step walkthrough of how to get ckb to connect/recongize to my K70 RGB.

isaacgr, 2016-07-26:
I am also having this issue. I got this driver in the hopes that it would fix my media keys not working in 16.04 but they still dont. Would appreciate any insight. This was marked as a solution for the k95 elsewhere but they dont mention how they came to the conclusion to use this fix. https://ubuntu-mate.community/t/volume-scroll-wheel-doesnt-work-in-16-04/5128/8

blip, 2016-07-31:
I just unboxed a K70 RGB. OSX 10.11.6, installed ckb 0.2.6. It doesn't detect the keyboard. I tried plugging in via my Monitor's USB hub and directly into the Mac. CKB says no device connected.

isaacgr, 2016-07-31:
So I should have said that I was using a VM. I tried using it with an actual install and it worked fine. I cant speak for any issues with the RGB, however. @blip Do you get any readings for the keyboard and buttons when using evtest?

blip, 2016-07-31:
@isaacgr sorry I'm brand new to this. What is, where do I find evtest? I am not using a VM (I tried, and I can't find any obvious way to pass thru USB/keyboard devices into a VM in VMware Fusion).

I plugged  into a Windows laptop, K70 Lux works great. Updated firmware to 2.0.5. Get back to Mac, nothing (plugged in directly into Mac).

I don't mean to hijack the thread if this is something different, hardware is recognized in the System Report:

Corsair Gaming K70 LUX RGB Keyboard :

  Product ID:	0x1b33
  Vendor ID:	0x1b1c  (CORSAIR MEMORY INC.)
  Version:	2.05
  Serial Number:	1600D03AAEB980E055C98039F5001943
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Corsair
  Location ID:	0x14100000 / 12
  Current Available (mA):	1000
  Current Required (mA):	500
  Extra Operating Current (mA):	0

$ sudo /Applications/ckb.app/Contents/Resources/ckb-daemon 
    ckb: Corsair RGB driver beta-v0.2.6
[I] Root controller ready at /var/run/ckb0

That's all I ever get. Plug and unplug, keyboard only plug, or both, direct into the computer or via Monitor USB hub, etc. Nothing happens past it saying it's ready.

tatokis, 2016-07-31:
First of all, K70 RGB means nothing, there are three different K70 RGB 
models.
A search on the bug tracker reveals plenty of info 
https://github.com/ccMSC/ckb/issues/386

Need to compile the newdev branch from the repository as that includes 
fixes for K70/K65 Lux/Rapidfire.
There are no prebuilt binaries for these keyboards yet, so compiling is 
the only option.
You can find info on how to do it on the README.

isaacgr, 2016-07-31:
@blip My apologies I read it and kept thinking it was a linux install.

blip, 2016-07-31:
@tatokis you're right K70 RGB means nothing. Yesterday I did not know that. Today I do.

I returned the K70 RGB [LUX] mainly because I didn't really like the font. The place I got it from only has that LUX RGB version so I was stuck. I downgraded to the K70 non-RGB version. I keep a backlight which is what I really wanted, and the font is easier on my eyes. Chatted with someone else they said their Razer is having some issues so it seems like it's a good idea to stick with Corsair. This version detected instantly. Key re-map works. Detects in ckb as K70R. So I think I'm getting what I really wanted out of it.