ccMSC/ckb #427: K65 Rbg lux doesnt show up

Xatl, 2016-07-28:
I dont know how to code or anytthing like that i need help configuring the program to work on my k65 lux because it isnt showing up on the program. ive looked for an awnser everywhere, if anyone knows how to fix it and has a skype add me : rebeusisgoodaf i really need help, i dont want to have to return this keyboard, On mac

SeanMcG, 2016-07-28:
Possibly it has a USB device ID that ckb doesn't recognize yet.I'm not sure how to get that information on Mac. Maybe is there a programs in /Applications/Utilities that can list USB devices?                                                                                                                                                                                                                                                                                           From:notifications@github.comSent:July 28, 2016 17:55To:ckb@noreply.github.comReply-to:reply@reply.github.comSubject:[ccMSC/ckb] K65 Rbg lux doesnt show up (#427)  I dont know how to code or anytthing like that i need help configuring the program to work on my k65 lux because it isnt showing up on the program. ive looked for an awnser everywhere, if anyone knows how to fix it and has a skype add me : rebeusisgoodaf i really need help, i dont want to have to return this keyboard, On mac

tbenz9, 2016-08-13:
@Xatl have you tried the newdev branch yet?  The RGB LUX is only supported on the newdev branch.