ccMSC/ckb #431: K65 & macOS Sierra

thb003, 2016-08-01:
I've got a Corsair K65 RGB Rapidfire but I can't use the cab app with it. I've done all the support you're proposes but anything happened, the "device is disconnected". When I look at my USB ports my Mac recognises as a K65 but not the App. On Windows (Bootcamp Windows 10) the Corsair Utility matches with the app. 
Does it posible to have more informations about the origin of this problem or how to fix it ? 

----------
MacBook Pro 15" Retina mid 2014
macOS Sierra 10.12 built 16A254g (latest)
Corsair K65 RapidFire 
CKB versions used : 0.2, 0.4, 0.5, 0.6

TwoD, 2016-08-12:
Which firmware version is in the keyboard? Have you upgraded it using Windows? Maybe ckb can't recognize it because of a newer firmware?

thb003, 2016-09-05:
Hi, 
I've updated the keyboard firmware with windows In fact ...