ccMSC/ckb #432: ckb wont install on debian MX-15

tjw0lf, 2016-08-02:
Can you help me please thank you

tjw0lf, 2016-08-28:
Can somebody help please?  26days no answer

fleischie, 2016-09-06:
Hi @tjw0lf 

Sorry if there is some delay sometimes. I suppose almost everybody is doing this on their free time, and considering it was summer on the northern hemisphere, it is possible everybody was on vacation. 😉 

But let me try to help you. 😃

(Let me start with the basics, as I have no previous experience with Debian MX-15):
- How do you install `ckb`: Is it via `ckb`? (I read MX is the antiX OS based on Debian 8.5 (Jessie))?
- I just read MX-15 advertises with being `systemd-free`. Could you run `which systemd` on the terminal just to make sure it is installed/it is not installed.

If we can make sure, that `systemd` is not installed, we might just know where the problem is. That in terms would make it easier to work around this problem. But first things first. 😉