ccMSC/ckb #433: CapsLock disable

t3kg33k, 2016-08-02:
I'm having a hard time finding a way to disable the CapsLock in the software. Is that not an option in the software?

fleischie, 2016-09-06:
Hi @t3kg33k.

Do you want to completely disable the key (as in: do not produce any input) or do you just don't want to enable the Caps Lock mode?

There is an option to rebind Modifier Keys:
[image]

I cannot safely say, that this will behave the way you intend it to. Maybe you could elaborate a little more on what you are trying to achieve? Maybe I can dig some more into this. ☺

t3kg33k, 2016-09-07:
Have the option to completely disable the Caps Lock. Rebind to the Shift key would work too. Thanks.
I believe in the Windows version of the software for the keyboard there is an option to disable the Windows key. I was hoping there would be an option to do the same for the Caps Lock.