ccMSC/ckb #434: Strafe RGB Windows Key doesn't bind to any command

volcan96, 2016-08-08:
FW 2.0.5 on Mac OS 10.11.5

Left option/Windows Key key doesn't pass any command. Not a HW issue since it works fine as windows key in windows or if ckb is uninstalled and use default behavior (i.e. acts as left command key).

Have uninstalled ckb. redownloaded and built from latest master.

I ran 
sudo launchctl load /Library/LaunchDaemons/com.ckb.daemon.plist

then
sudo /Applications/ckb.app/Contents/Resources/ckb-daemon

    ckb: Corsair RGB driver beta-v0.2.6
[I] Root controller ready at /var/run/ckb0
[I] Connecting Corsair STRAFE RGB Gaming Keyboar at /var/run/ckb1
[I] Starting input thread for /var/run/ckb1
[E] loadrgb_kb (led_keyboard.c:211): Bad input header
[W] _start_dev (device.c:45): Unable to load hardware profile
[I] Setup finished for /var/run/ckb1

Cojomax99, 2016-08-08:
👍  have the same issue

jerett, 2016-09-20:
have same issue