ccMSC/ckb #435: K70 Lux RGB

stevenkendrick3, 2016-08-11:
Hey everyone... I know that there are already other discussions on this problem, but to be completely honest I am so lost. I am by no means a programer. I was hoping that someone could explain to me like i'm 5 how to fix this problem. Everything I have tried shows there are no devices connected. I am so sorry for my incompetence but it would mean the world to me if someone could help me.

asmyers, 2016-08-11:
Hi Steven, You'll need to provide a bit more information.  At the very least what operating system you're on and what's not working.

stevenkendrick3, 2016-08-11:
Right now i am running OS X Yosemite Version 10.10.5 . I have not gotten around to upgrading. If upgrading would help I am more than happy to do so. If there is anything else you need please ask. Thanks

asmyers, 2016-08-11:
Okay, but you still haven't told us the problem :)  I unfortunately won't be able to help much because my k70 LUX is only coming today, and I'm running Linux.  If this is the same problem you posted in the other thread about a possible solution is there.  You have to install from the linked zip file instead of the release version.  

https://github.com/ccMSC/ckb/archive/newdev.zip

stevenkendrick3, 2016-08-11:
Okay I will try that. Im sorry I wasn't clear. The reason I haven't said the problem is because I don't know what the problem is haha. All i know is that no matter what i try the application just says no devices connected.

stevenkendrick3, 2016-08-11:
I have tried this file before... I open it and honestly am confused. I can not read the "Read Me" because I don't not have a program to open it. I am so sorry I don't understand what to do.

stevenkendrick3, 2016-08-11:
I ran the pre install script. However, Im not sure what to do from here. I then tried to run quick install and this is what I got.
Last login: Thu Aug 11 14:20:43 on ttys000
Stevens-MBP-2:~ stevenkendrick$ /Users/stevenkendrick/Desktop/ckb-newdev\ 4/quickinstall ; exit;
readlink: illegal option -- f
usage: readlink [-n] [file ...]
Preparing build files...
No suitable qmake found.
Please make sure you have Qt5 installed.
Re-run this program with QMAKE=/path/to/qmake to select qmake manually.

asmyers, 2016-08-11:
It sounds like you need to install some dev tools.  Take a look at the installing from source part of the readme: https://github.com/ccMSC/ckb#os-x-installation

tatokis, 2016-08-11:
https://github.com/ccMSC/ckb#building-from-source explains everything.
And make sure to use the link with the zip from above.

stevenkendrick3, 2016-08-11:
Thanks guy for the help. Im so lost. I think ill just accept defeat on this one. Thanks tho

tatokis, 2016-08-11:
Install XCode from the App store, and run it until you see the 
"welcome" screen.
Then install Qt from this 
http://download.qt.io/official_releases/online_installers/qt-unified-mac-x64-online.dmg
Finally run the quickinstall script and it will take care of the rest.

ArchJMC, 2016-08-12:
its asking what folder i want to install the QT in.. what do i put there?

crazydart, 2016-08-12:
Just got a K70 and had to do this... worked great.  Was showing no devices.  This is a great project.  I dont know why Corsair does not want an OSX version.  I am a developer not a gamer, but I want to have a nice keyboard.

boatchum, 2016-08-27:
For some reason I cannot install QT5. and they will not offer further support to me. I have even tried on all the 3 mirrors provided from qt. Is there a workaround QT5? 
Keyboard is recognized, CKB is installed (it just doesn't show anything other than settings when app is open), and showing the default Red/White LEDs. So basically the install begins and ends with this error every time: Could not download archive: http://download.qt.io/online/qtsdkrepository/mac_x64/desktop/tools_qtcreator/qt.tools.qtcreator/4.1.0-0qtcreator.7z : Connection closed

Please help thanks.