ccMSC/ckb #439: K70 RGB LUX MAC

DeviousR, 2016-08-15:
I just bought a new k70 rob lux like 2 days ago and when I heard about this app I was excited. When I downloaded it from binary I wasn't able to use the app because it said no devices connected. I have tried everything that was on the website on how to fix this and still no luck. Right now I am on El Capitan. Can someone please help me?

horen20wash, 2016-08-16:
LUX keyboards and RapidFire are only supported  in newdev branch. You can checkout the branch and build on your system and install including com.ckb.daemon.plist as described in BUILD.md.   All LUX keyboards will work.

tunicwriter, 2016-08-18:
Along with what @horen20wash said, in my experience you absolutely have to update your firmware, as mentioned in the troubleshooting for OS X:

```
Please ensure your keyboard firmware is up to date. If you've just bought the keyboard, connect it to a Windows computer first and update the firmware from Corsair's official utility.
```

My K70 Lux RGB would not work, at all, on El Capitan until I updated the firmware (out of the box, it came with firmware v1.something. It's now on v2.05).

bwoodruff, 2016-08-24:
Yep -- can confirm: had to plug the keyboard into a Windows machine with CUE to update the firmware before ckb would recognize it.

cranky-mike, 2017-02-02:
I purchase a K70 LUX RGB a few days ago. The keyboard is firmware v2.05. I am running OS X v10.11.6. I've finally gotten it to work in VM Fusion by forcing the USB control to my Windows 7VM but in that case you can only use it in window. I've also researched as much as I can find on this website but it still will not recognize the keyboard. I installed the existing package and I also did a compile with the NewDev but still nothing. I don't see this exact model in the usb.h code (vendor code: 0x1b33 product code: 0x1b1c). 

I am new to this but I understand the process. Also when I compile from QT Creator there are a bunch of errors that show up but if I use "quickinstall" it does at least create the app. I have rebooted in between and plugged and unplugged it as well and it never sees the keyboard. Any help would be appreciated.

Thanks in advance.

darkwork, 2017-08-09:
Hi, just jumping on to this.  I just got the keyboard, pretty new.  The keys would not work til I am in Bios mode.  Then the keys like arrow or backspace works but the multimedia keys won't work.  But if i am in normal mode, like 1, then all multimedia keys work and letters work, but arrow or backspace won't.  Its super weird and messed up.  Anyone have the same issue and or have a solution for it for OSX 10.11.6?  Thanks.