ccMSC/ckb #447: Any new updates?

aDallo26, 2016-08-22:
I've been keeping an eye out for a while now, but have not seen any new updates come through. I'm using a K65 Lux RGB on a 5K iMac running El Cap v10.11.6 and the software still does not detect the keyboard. 

Any new updates?

batchpert, 2016-08-22:
Hi i have the Strafe RGB and i had  similar problem a few months a go. My solution was to download the testing branch download which seemed to solve the problem and my keyboard was detected straight away. 

I am also hoping for an update soon maybe even v0.3 ;P for the macros

aDallo26, 2016-08-22:
Where do I get the testing branch one?

TwoD, 2016-08-22:
@aDallo26 There are no official releases from the testing branch so you'll have to clone the git repo, check out the branch and compile it yourself.

horen20wash, 2016-08-27:
Looks like the project is slowly dying.

TwoD, 2016-08-28:
@horen20wash I doubt that. It's only been about two months since the last commit and developing software usually progresses at an uneven pace, if at all during some periods.

The maintainer could be on a long vacation, or having personal issues, or is simply too busy with other things right now. They aren't getting paid for their work on this, which of course factors in to how it is being prioritized. The best we can do, if we aren't able to investigate problems further or write patches ourselves, is simply to wait and encourage the developer(s) when they do get back to it.

I've tried to help out here every now and then by asking people to provide more information in their reports. Seeing a bunch of issues pile up without having enough information to properly diagnose them and make progress can be very frustrating and demoralizing, even to the point where you as a developer/maintainer feel very helpless and no longer feel able to get anything done because you can't keep up with the previous pace. It takes a while to get there, but the risk increases with the amount of backlog waiting on your input/feedback/diagnosis/decisions.

fleischie, 2016-09-06:
@horen20wash You can always donate towards the main dev/maintainer to ensure there is an incentive to work for. 😉

SicVolo, 2016-09-22:
@ccMSC are you keeping this alive? We can see you are back from vacation and playing Splatoon again :)