ccMSC/ckb #449: ckb needs restart after hibernation

olofson, 2016-08-24:
While upgrading to a 4.x Linux kernel solved #313 for me, there is one remaining issue, that results in the same apparent symptoms; that the keyboard would disconnect and reconnect every few seconds:
```
[50452.813771] usb 3-1.1.3: USB disconnect, device number 15
[50453.093749] usb 3-1.1.3: new full-speed USB device number 16 using xhci_hcd
[50453.187588] usb 3-1.1.3: New USB device found, idVendor=1b1c, idProduct=1b11
[50453.187592] usb 3-1.1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[50453.187593] usb 3-1.1.3: Product: Corsair K95 RGB Gaming Keyboard 
[50453.187594] usb 3-1.1.3: Manufacturer: Corsair
[50453.187596] usb 3-1.1.3: SerialNumber: 0A00E00DAEA7A44454E0C100F5001945
[50453.189410] input: Corsair Corsair K95 RGB Gaming Keyboard  as /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1.1/3-1.1.3/3-1.1.3:1.0/0003:1B1C:1B11.000D/input/input35
[50453.190145] hid-generic 0003:1B1C:1B11.000D: input,hidraw3: USB HID v1.11 Keyboard [Corsair Corsair K95 RGB Gaming Keyboard ] on usb-0000:00:14.0-1.1.3/input0
[50453.191636] input: Corsair Corsair K95 RGB Gaming Keyboard  as /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1.1/3-1.1.3/3-1.1.3:1.1/0003:1B1C:1B11.000E/input/input36
[50453.192229] hid-generic 0003:1B1C:1B11.000E: input,hiddev0,hidraw4: USB HID v1.11 Keyboard [Corsair Corsair K95 RGB Gaming Keyboard ] on usb-0000:00:14.0-1.1.3/input1
[50453.192307] usbhid 3-1.1.3:1.2: couldn't find an input interrupt endpoint
[50453.454206] input: ckb1: Corsair K95 RGB Gaming Keyboard as /devices/virtual/input/input37
[50453.455547] input: ckb1: Corsair K95 RGB Gaming Keyboard as /devices/virtual/input/input38
```

However, with Ubuntu kernel 4.4.0-34-generic, this only happens after waking up from hibernation, and I can "fix" it by just restarting ckb. (That didn't help with the 3.x kernel.)

I don't know if this impacts my other computer (the one where ckb has worked perfectly all along), as I never use hibernation on that one.

opsroller, 2016-08-30:
This can be fixed in the systemd unit file. I'll take a peek tonight.

olofson, 2016-09-20:
Bad news: I suddenly started having disconnect issues with this <censored> laptop again, for no apparent reason...

More bad news: Figured I'd use hardware macros with the standard driver instead - but those are not supported on K95 RGB!

Good news: Actually running an animation at 30 fps seems to eliminate the disconnects. o.O May or may not be a hint as to what's going on here.

frickler24, 2016-09-23:
@olofson For the G-keys there is a macro binding service in the GUI. Please use the testing branch (up to now it is not yet released) or use an improved version f7f8ce9 from my fork at https://github.com/frickler24/ckb.git. the current branch is macrotime.0.3

olofson, 2016-09-24:
@frickler24 I'm not sure I'm following... Have you implemented hardware macros somehow?

The problem I've been having is that I couldn't use ckb at all on that machine, because it causes the keyboard to disconnect and reconnect regularly, causing issues such as stuck keys, random keyboard layout switching etc, apart from the keyboard simply being dead for a few seconds every now and then.

As for the G keys, I actually prefer the "Program" binding method (with wmctrl) over macros, as that avoids interference with keyboard input in any applications. I was just looking for a solution/hack to have the G keys work without ckb, like I did earlier, with the K90 keyboard.

Anyway, somehow, running a LED animation seem to avoid the disconnect issue somehow. Will investigate that further at some point. (I don't really need or want animations. I just want the keyboard - and the G keys - to actually work reliably.)