ccMSC/ckb #451: how to change brightness via /dev/input/ckb*/cmd

lukedoomer, 2016-08-29:
whats the correct command to change brightness?
because [https://github.com/ccMSC/ckb/blob/master/DAEMON.md] has no such information

lukedoomer, 2016-09-02:
I believe there is a way because we can set brightness from ckb-gui then ckb-gui is capable of passing the changes to the daemon.
I try to trace the code and still got no answer.
I just need to control the brightness from the command line.
HELP!

fleischie, 2016-09-03:
Hello @lukedoomer do you want to set the general lighting for the whole keyboard?

I looked at the ckb-gui to see where the dimming is set and I found it in [src/ckb/kblight.cpp#L223](https://github.com/ccMSC/ckb/blob/testing/src/ckb/kblight.cpp#L223):

`cmd.write("rgb <keyname>:<rgb-color>")`

So, I would suggest you try to write `rgb 000000` to the ckb-daemon to test, whether this works globally (or `rgb ffffff` depending on your current dimming). And then `rgb enter:ff00ff` or similar for particular keys.

(The list of key names can easily be looked up here [src/ckb/colormap.cpp#L10](https://github.com/ccMSC/ckb/blob/testing/src/ckb/keymap.cpp#L10). It's always the third element per keymap-item. 😉 )

Hope it helps and you can resolve your issue. 😃 

P.S.: I would have liked to test it out myself and be more definitive but I don't have my keyboard here currently.

lukedoomer, 2016-09-05:
@fleischie From the interface, we can set the brightness to 33% 67% 100%. rgb command is able to set color of a specific key or the whole keyabord and is not able to set the brightness. ( I guess not because manual of the driver daemon doesn't mention it.)
Thanks!

fleischie, 2016-09-05:
@lukedoomer I took the liberty to attach my K70, so I can test. :wink:

I look into the [`DAEMON.md` under the LED Commands section](https://github.com/ccMSC/ckb/blob/master/DAEMON.md#led-commands). It states mostly the same as I checked from the `ckb` GUI sources. Additionally you can set more than one key to a specific color.

But I now tested and I was able to dim the keyboard with the following sequence of commands:
1. `echo active > /dev/input/ckb1/cmd` (enable the keyboard to take commands via the daemon).
2. `echo rgb ffffff > /dev/input/ckb1/cmd` (set the whole keyboard white)
3. `echo rgb 666666 > /dev/input/ckb1/cmd` (set the whole keyboard to a dimmed white)
4. `echo rgb f2415f > /dev/input/ckb1/cmd` (set the whole keyboard to a hot pink 😉 )
5. `echo rgb d1304d > /dev/input/ckb1/cmd` (set the whole keyboard to a dimmed hot pink)

If I take this specific routine in [`src/ckb/kblight.cpp` line 347](https://github.com/ccMSC/ckb/blob/testing/src/ckb/kblight.cpp#L347) it calculates the color of each component (red, green, blue) makes a linear value out of it (a value between 0 and 1) multiplies this value by the dimming-value and makes an sRGB value out of it again (0 - 255).

<details>
The dimming value is inverted and linearized with the followin:
`float light = (3 - _dimming) / 3.f;`
with `_dimming` a value between 0 (100%) and 3 (off).

It then create a `qRgb` value for the QT application, which is an ARGB value, but only with these three components (r, g, b), (which further calculates a rgba value of (r, g, b, 255). So I guess we can ignore this conversion. 😊
</details>

So in general it should be possible to set each key to a specific dimmed color, but it requires some steps of calculations. 💻 

I hope you still can solve this problem. :smile:

lukedoomer, 2016-09-09:
thank you for your explanation. I know what to do now. so the dimming value is calculated to the brightness level.

closed 2016-09-09 by lukedoomer