ccMSC/ckb #452: two issues of K65 RGB in macOS 10.12 developer Beta 8

owenzhao, 2016-08-30:
Everything works fine in macOS 10.11.6.  After upgrading to 10.12 developer Beta 8. I find two issues.

1. No more keystroke repeating. If I press and hold some key, say "F", there is only one "f", instead of "ffffffff...".
2. "Shift + left click" to select text stops working. I use "Shift + left click" to select long text a lot. To test if it dues to the OS, I opened a macOS 10.12 beta 8 vm in Parallels Desktop 12. It turned out that "Shift + left click" worked in the VM. So I guess maybe it is related here.

USB Device ID: 0C01001BAEA7A44454E0F26EF5001947
Poll rate: 1 ms
Firmware version: 2.05
ckb version: beta-v0.2.6

owenzhao, 2016-08-30:
I cloned back to macOS 10.11.6. And I found out the repeat doesn't work in 10.11.6 either. It works in input method like pinyin, but not work in English (the default U.S. setting). 

If the keystroke was bridge by input method, the repeat works. It only doesn't work with the default U.S layout.

raparkhurst, 2016-09-03:
I can confirm that "Shift + Left Click" doesn't work in the latest Mac OS "Sierra" beta (16A312a).  I have the RGB K95.  I am using the latest ckb release for Mac OS (0.2.6 beta)

parralix, 2016-09-08:
i confirm the same. Shift click does not work, command click doesnt work either. k95, using ckb 0.26 on sierra GM

owenzhao, 2016-09-08:
It is said by the developer of Karabiner, in macOS 10.12, the keyboard driver is rewritten by apple. So it will take long time to update the app to the latest 10.12.

If that is true, I think we should give more time to ckb too. I personally will not upgrade to macOS 10.12 until this is fixed.

patmccormick, 2016-09-12:
OS X Sierra 10.12 beta (16A312a): Everything works except shift-selecting multiple items and control-selecting multiple items as indicated by others.

Although the driver may be rewritten, it should be a fairly simple task to find and correct the handling of these 2 modifier keys.

Otherwise everything else has been operating properly.

patmccormick, 2016-09-26:
It must be the part of the code that watches when a key is pressed & modifies behavior based on the use of control key modifiers; control, option, shift, etc., and also watches the duration that the keys are held down.  This program seems to be finishing its operation each time a single key is pressed without waiting for the rest of these possible combinations & situations.

jianghaizhi, 2016-11-16:
I met the same problem. "Shift + left click" to select text doesn't work