ccMSC/ckb #453: K70 Lux not working when CKB is running

fgblomqvist, 2016-09-02:
I edited the usb.h file so that CKB would recognize my new K70 Lux (no RGB or rapidfire, just cherry red), but once it did recognize it the keyboard just stopped working. I fixed it by killing the CKB daemon using my laptops keyboard.

I have updated the keyboard to the latest firmware through Windows, and I'm running Ubuntu 16.04, vanilla kernel.

I'm guessing this is simply because CKB is not made to support this keyboard yet?

(Side note: if I use my volume scroll, when CKB is not running and my keyboard is working, it crashes Unity)

fgblomqvist, 2016-09-03:
I tried the newdev branch as well but same thing happens.

horen20wash, 2016-09-03:
What is the vendor/product code of your keyboard (K70 LUX no RGB).

fgblomqvist, 2016-09-03:
0x1b36

horen20wash, 2016-09-03:
You added it to usb.h with a label and updated usb_linux.c or usb_mac.c with the label. Do you see any output when you start  it from the command line ?

fgblomqvist, 2016-09-03:
Sorry I meant that I replaced another keyboard:
```
#define P_K70_NRGB      0x1b36
#define P_K70_NRGB_STR  "1b36"
```
I chose to replace the K70 NRGB since mine is non-RGB and K70. Maybe I should try a LUX instead.

fgblomqvist, 2016-09-03:
After some advice from the MEGATHREAD I replaced the K70 rapidfire instead, and now it works!
This is how it should look in usb.h:
```
#define P_K70_RFIRE     0x1b36
#define P_K70_RFIRE_STR "1b36"
```

closed 2016-09-03 by fgblomqvist

rkingsbury, 2016-09-21:
@fgblomqvist I just purchased a K70 Lux (non-RGB) and am trying to get it to work. Did your keyboard work with the standard USB driver, or do you HAVE to have ckb working to recognize it?

Running Mint 18 with the stock (Ubuntu 16.04, 4.4.0-21) kernel, my keyboard is unusable. I've updated to firmware 2.04 and added the USB quirks to the command line but that doesn't help. In fact, after the firmware update, I can't even boot my system with the keyboard plugged in.

Was this your experience, too?

fgblomqvist, 2016-09-21:
@rkingsbury see my response in the other thread.