ccMSC/ckb #454: K70 Lux RGB Mac Mute Key

mwilliams17, 2016-09-05:
I have the ckb app running, and I can change animations, create new combinations, etc.

However, the mute key on my keyboard does not light up when I mute the sound. It did light up when I was downloading firmware through CUE on a windows computers. Wondering if this is something that is going to be added in to the ckb.

Thanks

fleischie, 2016-09-05:
I cannot safely say, as I do not have a K70 Lux RGB (only a K70), but under the Performance tab:

[image]

There is the `Indicate mute:` option in the top-row fieldset, which you can set lights for the mode on/off/unknown:

[image]

mwilliams17, 2016-09-05:
Thanks for the reply, I went and played with the settings in the "Performance" tab for the mute button.

Turns out the button is in the "Unknown" state when I check the box next to "Indicate mute". I figured this out by changing the color of the "Unknown" state when the box was checked, and sure enough, the color of the mute button matched the "Unknown" color.

I also changed the colors of the "On" and "Off" buttons for the "Indicate mute" section. The button color does not change when I press it on and off.

Thanks again for your help. Now at least I know the problem, and I can turn the mute button light on, even if its always on regardless of whether or not the sound is muted.

fleischie, 2016-09-06:
Hi @mwilliams17 again.

I am knee-deep in the source code looking for why the status can be unknown. 😀 

Do you have OS X or Linux?
<details>I am asking, because there are different ways to figure this out on these systems.</details>

## Mac _(which I am personally not as familiar with)_

It seems as if the default API is used to query the Audio Device. I honestly have not figured out how to navigate the docs on the Apple site, so sorry for that. But if that does not work and you are on a Mac, then we have to ask someone more experienced than me. 😉 

## Linux _(my go-to OS :wink:)_

In linux the `pacmd` command is used to query the pulseaudio sound server. Which may not work correctly, if pulseaudio is not installed.