ccMSC/ckb #456: ALL lights are gone after CKB

novicean, 2016-09-06:
The basic lights (red and white color) are appeared on keyboard when my mac is turning on.
But after log in, basic light (red and white color) is turned off.
(CKB is automatically turned on when I log in OS X)
Nothing is changed with CKB profile (M1, M2 and M3).
The lights revive if i quit CKB from status bar.
How can i fix this?

I use OS X 10.10.5, K95 RGB, CKB 0.2.6.

fleischie, 2016-09-06:
Hey @novicean 👋😄

I suppose the lights are on by default, i.e. the hardware profile, that is stored onto the device itself.

- Did you have this problem for long?
- When did this start to occur?
- Did you try to use the hardware button to dim the keyboard? Does this change anything?
- What is the default profile, that should be displayed?

Is the daemon running in the background? (Run `ps -ef | grep ckb` in a terminal, there should be a line reading `ckb-daemon`.) Normally the daemon controls the colors/animations/effects, that cannot be stored onto the device with the `ckb` software.

If you can provide more information, I am sure we can fix this. 💪 😃

novicean, 2016-09-07:
Thank you for your reply @fleischie . 
Here is more information.
- Did you have this problem for long?

-> I installed CKB yesterday. It worked well for short time. The Rainbow effect was beautiful. But a couple of minutes later, All lights were dimmed off.
- When did this start to occur?

-> 5 minutes later from first installation.
- Did you try to use the hardware button to dim the keyboard? Does this change anything?

-> No. Nothing is activated pressing hardware button
- What is the default profile, that should be displayed?

-> The default profile is CKB demo. It doesn't work on my keyboard.
- Is the daemon running in the background?

->Yes. It looks like that.
> DHANs-iMac:~ DHAN$ ps -ef | grep ckb
  501   332     1   0  8:26PM ??         0:00.89 /Library/PreferencePanes/TrackballWorks.prefPane/Contents/Resources/TrackballWorksHelper.app/Contents/MacOS/TrackballWorksHelper
  501   380     1   0  8:26PM ??         3:41.62 /Applications/ckb.app/Contents/MacOS/ckb --background
    0  1208     1   0  8:55PM ??         0:00.01 sudo /Applications/ckb.app/Contents/Resources/ckb-daemon
    0  1209  1208   0  8:55PM ??         0:02.42 /Applications/ckb.app/Contents/Resources/ckb-daemon
  501  1242   380   0  9:00PM ??         0:00.15 /Applications/ckb.app/Contents/Resources/ckb-animations/ckb-wave --ckb-run
  501  1243   380   0  9:00PM ??         0:00.15 /Applications/ckb.app/Contents/Resources/ckb-animations/ckb-gradient --ckb-run
  501  1251  1247   0  9:01PM ttys000    0:00.00 grep ckb

And I think you need these also.

1. The lights are turned on again if i kill CKB manually. But the colors are Just red and white(basic setting)
2. Firmware version is 2.05.
3. Driver log is as below

> ckb: Corsair RGB driver beta-v0.2.6
[I] Root controller ready at /var/run/ckb0
[I] Connecting Corsair K95 RGB Gaming Keyboard  at /var/run/ckb1
[I] Starting input thread for /var/run/ckb1
[E] loadrgb_kb (led_keyboard.c:211): Bad input header
[W] _start_dev (device.c:45): Unable to load hardware profile
[I] Setup finished for /var/run/ckb1

Thank you again.

fleischie, 2016-09-07:
OK, just to understand this correctly:
- It worked once, with a fresh install for 5 minutes.
- It never worked again? (Or does it periodically stop working?)

Have you tried to compile different branches (master/testing/newdev) and run the program again? Maybe there was a change lately, that fixes this issue?

I unfortunately have neither OS X, nor the K95 RGB, but I will do my best to help. 👍  😄

novicean, 2016-09-08:
Thank you @fleischie .

- It worked once, with a fresh install for 5 minutes.

-> Yes. Just 5 minutes.
- It never worked again? (Or does it periodically stop working?)

-> No. It never worked again.

and i tried to compile all 6 branches and ran the program.
But Nothing was happened for each case.

Do you need other information?

fleischie, 2016-09-09:
@novicean Hmm. Could you specifically run only the daemon?

So something in the form of:
- `sudo killall ckb; sudo killall ckb-daemon` - Stop everything.
- `sudo ./bin/ckb-daemon` - It should run in the foreground, now.

And then check, whether the daemon accepts input and acts accordingly:
- `echo rgb ffff00 > /var/run/ckb1/cmd`
- `echo rgb 00ffff > /var/run/ckb1/cmd`
- etc.

Only to see, whether the problem stems from the daemon?

As the log suggests, there is an issue with the `led_keyboard.c` saying "Bad input header", which originates from reading the colors, when loading the hardware profile. I will have to do some additional c-code-reading to understand exactly what's going on in there and whether this could affect the behaviour you are describing. Today I have no specific understanding of the daemon, but I wanted to report back to you, so you know that (although I do not have that much time) I am giving my best. 😄

novicean, 2016-09-09:
Hi. I always appreciate to your reply @fleischie .

> sudo killall ckb; sudo killall ckb-daemon
sudo ./bin/ckb-daemon

I entered first command, But I couldn't find "./bin/ckb-daemon"
So, I re-started "ckb.app" by clicking.
And then, When I entered below commands,

> echo rgb ffff00 > /var/run/ckb1/cmd
echo rgb 00ffff > /var/run/ckb1/cmd
etc.

My K95 dimmed shortly with some colors (ff0000 was red, 00ff00 was green).

On the other hand, I wonder why `Bad input header` message was appeared.
In `led_keyboard.c`, I found the loop which print that error message.
In first loop, `in_pkt[0]` and `data_pkt[0]` have same value (maybe `null`).
But in second loop, `in_pkt[1]` was filled with garbage.
I printed the value adding below codes above `ckb_err("Bad input header\n");`
`printf("in_pkt[%d] : %s \ndata_pkt[%d] : %s \n", i, in_pkt[i], i, data_pkt[i + clr * 4]);`

the result is as below.
`in_pkt[1] : ?<?????????????????????????????????????????????????????????????<`
`data_pkt[1] : ?<`

As I told you, I really thank to you, not only your help but kind reply.
I don't want to get more information from you, if you have busy days.
Have a nice day!

fleischie, 2016-09-09:
Maybe the structures were filled with binary data, that cannot be correctly converted to prinable characters. That's why there are so many question marks. 😃 

OK, sorry I didn't know OS X is that much different from unix. Check the files in `/Applications/ckb.app/Contents/ckb-daemon.log`. Maybe there is more data there, that can be helpful.

Because, if the keyboard takes in commands and works appropriately for a specified amount of time, then why is it not working after a few seconds?

Thank you @novicean , have a nice day, too. 😄