ccMSC/ckb #457: All lights turning off on Arch / STRAFE RGB

tralce, 2016-09-07:
I've had this happen twice now. I'd love to send you any information you might need.

I'm running the latest ckb-git on the latest Arch. Lights go out suddenly on the STRAFE RGB. killall ckb and systemctl restart ckb-daemon accomplish nothing. Reboot, nothing. Only way I've been able to recover my lights is to pacman -Rsn ckb-git, rm -r ~/.config/ckb then remake my config from scratch. 

Are there any logs or anything I can send you? Here's a pastebin of my ckb.conf:

http://pastebin.com/bDaZLRWu

fleischie, 2016-09-07:
Hey @tralce 

From two minutes of research it seems, that the `ckb-git` AUR package is on a specific version, that might just not have included the appropriate patches to support your keyboard.

- You can try, whether `ckb-git-latest` works out of the box (making sure the daemon is enabled).
- You can always clone the project, checkout the testing branch and compile for yourself.

Additionally I haven't track the support of newer devices (as I have a K70 RGB, which was fortunately always supported), so maybe someone with more experience in this regards could also check? 😃

NovaCygni, 2016-10-06:
Had this issue with my K70 awhile ago, it came down to a USB power saving setting in which the keyboard was loosing power when not in use thus all the lights would go out...