ccMSC/ckb #458: Support for older K65 (Non RGB)

JoeSayerUK, 2016-09-07:
I see it is not listed as supported but has anybody been able to use CKB with the older K65 units without backlighting?

Currently the keyboard is pretty unusable on El Capitan unless in BIOS mode which still leaves a few unresponsive keys.

Any ideas?

cb-fred, 2016-10-06:
I'd like to know as well.

cb-fred, 2016-10-09:
I have re-compiled the daemon adding the older k65 device id `0x1b07` to the device list and got it to recognise the keyboard. Unfortunately the keybord then completely stops from working. 

[patch.txt](https://github.com/ccMSC/ckb/files/518335/patch.txt)

cb-fred, 2016-10-09:
I have updated the patch to label the keyboard as non-rgb. The UI doesn't seem to recognise it anymore, but the keyboard seems to work. 

It also seems that I had some weird conflicts with UsbOverdrive and Karabiner, which I have uninstalled. Maybe that did the trick as well. 

[patch.txt](https://github.com/ccMSC/ckb/files/518347/patch.txt)

cb-fred, 2016-10-09:
Here's the build if anybody wants to try.

[ckb.zip](https://github.com/ccMSC/ckb/files/518364/ckb.zip)

wqfan, 2016-12-02:
Not sure if you guys are still following this thread. I got a non-RGB K65 yesterday and found the Caps Lock light didn't work on macOS 10.12.1. I tried the ckb and had no results. However, I found an application called [Karabiner-Elements](https://github.com/tekezo/Karabiner-Elements) by which the Caps Lock light now works well and lots of keys can be remapped. Hope this can help you resolve your problems!