ccMSC/ckb #461: GUI freezes on opening color chooser

Trollwut, 2016-09-09:
Hi folks!

I have `ckb` installed for only a few days, but since first launch, this issue happened.

The GUI works as intended, but when I select keys and click on "Change color...", the GUI freezes completely and has to be killed.

The color chooser never worked, so unfortunatelly the "replicate error" for me is to install the software and press on "Change color...".

I'm using Arch with GNOME, everything up-to-date.

fleischie, 2016-09-09:
Hello @Trollwut  👋 

- Did you run the commands (`ckb-daemon` and `ckb`) on the terminal?
- Did it produce any alarming output (warnings/errors/segfaults)?

Could you try out to only run the daemon and interact with it on the command line just to see, that it's really the gui, that breaks? (From the project's main directory):
- `sudo ./bin/ckb-daemon`
- `echo rgb f:00ffff > /dev/input/ckb1/cmd` (or something similar)

This could tell you, whether it is possible to change the color of a single key (for example the `f` key) to a specific color. If this is not possible, we have to dig deeper. 😉

Trollwut, 2016-09-09:
Servus!

First I stopped/quit all software, of course.

In one terminal, I startet `sudo ckb-daemon`, which seems to work as intended. No errors, just normal output:

```
$ sudo ckb-daemon

    ckb: Corsair RGB driver beta-v0.2.6
[I] Root controller ready at /dev/input/ckb0
[I] Connecting Corsair K70 RGB Gaming Keyboard at /dev/input/ckb1
[I] Starting input thread for /dev/input/ckb1
[E] loadrgb_kb (led_keyboard.c:211): Bad input header
[W] _start_dev (device.c:45): Unable to load hardware profile
[I] Setup finished for /dev/input/ckb1
```

Second terminal for `ckb`. At opening, I STRG-selected WASD and pressed on "Change color...", which immediately made the app freeze. Killed because of reasons. This is the console output:

```
$ ckb
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
Scanning  "/opt/ckb/ckb-animations/ckb-gradient"
Scanning  "/opt/ckb/ckb-animations/ckb-pinwheel"
Scanning  "/opt/ckb/ckb-animations/ckb-rain"
Scanning  "/opt/ckb/ckb-animations/ckb-random"
Scanning  "/opt/ckb/ckb-animations/ckb-ripple"
Scanning  "/opt/ckb/ckb-animations/ckb-wave"
Starting  "/opt/ckb/ckb-animations/ckb-wave"
Starting  "/opt/ckb/ckb-animations/ckb-gradient"
Downloaded new firmware list. 6 entries found.

(ckb:30148): GLib-GObject-WARNING **: cannot register existing type 'GtkWidget'

(ckb:30148): GLib-GObject-WARNING **: cannot add class private field to invalid type '<invalid>'

(ckb:30148): GLib-GObject-WARNING **: cannot add private field to invalid (non-instantiatable) type '<invalid>'

(ckb:30148): GLib-GObject-CRITICAL **: g_type_add_interface_static: assertion 'G_TYPE_IS_INSTANTIATABLE (instance_type)' failed

(ckb:30148): GLib-GObject-WARNING **: cannot register existing type 'GtkBuildable'

(ckb:30148): GLib-GObject-CRITICAL **: g_type_interface_add_prerequisite: assertion 'G_TYPE_IS_INTERFACE (interface_type)' failed

(ckb:30148): GLib-CRITICAL **: g_once_init_leave: assertion 'result != 0' failed

(ckb:30148): GLib-GObject-CRITICAL **: g_type_add_interface_static: assertion 'G_TYPE_IS_INSTANTIATABLE (instance_type)' failed

(ckb:30148): GLib-GObject-CRITICAL **: g_type_register_static: assertion 'parent_type > 0' failed

(ckb:30148): GLib-GObject-WARNING **: cannot add private field to invalid (non-instantiatable) type '<invalid>'
Killed
```

On the other hand, starting the daemon in a terminal and `echo rgb s:ff00ff > /dev/input/ckb1/cmd` didn't produce any problem (nor did it change the color of a key), but opening `gedit /dev/input/ckb1/cmd` tells me that "/dev/input/ckb1/cmd is not a regular file."

Does this help you for troubleshooting?

tatokis, 2016-09-09:
Which version of GLib do you have installed?

I ran ckb on a terminal and I got none of these GLib warnings, so I 
think they might've changed something that broke ckb, or even worse, 
Qt. (Although not sure how.)

Trollwut, 2016-09-09:
I don't know for which package I should look for, these are the two that are looking most promising:

```
core/glib2 2.48.2-1
    Low level core library
core/glibc 2.24-2 (base)
    GNU C Library
```

fleischie, 2016-09-10:
Moin! 😉 

I have installed both `glib2` and `glibc` and it works for me. Perhaps there are issues between the gtk-environment of GNOME and Qt?

tatokis, 2016-09-11:
What happens if you run ckb with different theme engines?

For example, make sure it is closed and then run it like this
QT_STYLE_OVERRIDE=adwaita ckb
and
QT_STYLE_OVERRIDE=gtk ckb

If you have any other engines installed, try them as well, and see if 
any of them fix the issue.
I suspect it is an issue with the default one, since the application 
uses Qt, but the warnings are from Gtk.

Trollwut, 2016-09-12:
> QT_STYLE_OVERRIDE=adwaita ckb
> and
> QT_STYLE_OVERRIDE=gtk ckb

Unfortunatelly the same errors as in the initial post. :/

Sorry, I don't have any experience about theme engines, so I can't compute any of these lines.
Just to tell again: EVERYTHING works as intended, just the color picker doesn't.

fleischie, 2016-09-12:
Do you have another application, where the Color-Chooser works? Preferably a QColorDialog-based one, but I would like to know, whether the GTK-based Color-Chooser widget works on your setup?

The sources state, that a click on the button triggers the `QColorDialog::getColor()` method. The freezing could be an issue of the QColorDialog opening in a GTK environment.

Trollwut, 2016-09-12:
AFAIK this is the only Qt app I'm using. May you please commend me a little one with color chooser, so I can test it?

fleischie, 2016-09-13:
Hey @Trollwut 

I did some research regarding Qt applications, I found the following link: [Arch Linux Qt Wiki](https://wiki.archlinux.org/index.php/Qt#GTK.2B_and_Qt). It could be helpful in setting the styling between gtk+ and Qt applications.

You can download the example standard QtWidgets [here](http://doc.qt.io/qt-5/qtwidgets-dialogs-standarddialogs-example.html). I wish I could think of a simpler method to test this. 😐

<details>
A short description on how to use this example:
- Download/Paste the four files separately into the same directory.
- (make sure `qmake` is installed and) run `qmake` 
- run `make`
- select the "Color Dialog" item in the toolbox
- try out the options in the bottom of the widget
- if it works, we have to check something else 😄 
</details>

Trollwut, 2016-09-14:
Works flawlessly with any setting.

(Well, non-native with *no* buttons can't accept the dialog, but it opens, so... guess it works.)

Trollwut, 2016-09-14:
Tested it again in `ckb`, but still freezes.

If I select one button, the color chooser still says that the selected color is `255, 255, 255`. Is this normal behaviour?

[Cool, an image!](http://i.imgur.com/1jvfcOW.png)

fleischie, 2016-09-14:
@Trollwut i guess not. 😋

Can you open the color chooser dialog in the performance settings-tab. For example the color of the brightness button on the different levels?

Trollwut, 2016-09-14:
Nope, also freeze...

Trollwut, 2016-09-16:
Ok, I reinstalled my system, as I wanted to do this anyway, so sorry that I can't provide a solution.

It now works wonderfully. Still a GTK DE (XFCE), so I guess any Qt package has been... modified or missing or whatever. At least it works now. :)

fleischie, 2016-09-17:
Glad it worked out in the end. :smile: :+1:

Trollwut, 2016-09-24:
Yessir! Played a bit around - nice piece of software!

DenyBoy, 2016-11-10:
Have sam issue. This makes it work for me.

QT_QPA_PLATFORMTHEME=gtk2 ckb