ccMSC/ckb #463: Corsair K65 Lux RGB Compatibilty

PopTart99, 2016-09-12:
I have tried everything from downloading the package and using qt to try to build this program for my rgb and all that is showing when I have the app open is the device is not connected or on Qt it has many issues. Everything has failed to work for me and need help because I only have a macbook air and am trying to run this program. Also this mac is being ran on OS X el captain 10.11.6 . Please send help as soon as possible.

horen20wash, 2016-09-12:
Did you build ckb from the newdev branch ?

PopTart99, 2016-09-12:
Yes I did

horen20wash, 2016-09-12:
Please review the usb.h file if the Device id from the OS of your keyboard is in the file.

horen20wash, 2016-09-12:
Few observed also, that they had to run a firmware update before it was detected, I did not have that problem. 
Please run the firmware update only on a Windows system.

PopTart99, 2016-09-13:
what's the fw update and where is that?

PopTart99, 2016-09-13:
and how do i find Device ID

fleischie, 2016-09-13:
@PopTart99 Hey.

The fw-update refers to the official CUE software on Windows. It suggests to update the firmware for you and is the official way to update the firmware.

For finding the device ID see this thread: [MEGATHREAD: Newer keyboards/mice](https://github.com/ccMSC/ckb/issues/386).