ccMSC/ckb #464: Keyboard mouse commands

horen20wash, 2016-09-14:
This is on MacOS.
Objects can be grouped together by holding down the command key and the mouse pointer. This operation in not supported at all with the ckb driver. Same applies to shift and mouse.
Any details into this would be appreciated.
Thanks

Thomas-Z, 2016-09-16:
Just want to add that (at least for me) it is working perfectly fine with OS X 10.11.6 but not anymore with macOS Sierra (10.12).
Command + Click and Shift + Click selections don't work anymore.

I tried without the driver and it does not work either so I'm not sure what might be the problem.

Thanks for your great work !

horen20wash, 2016-09-16:
Thanks for the update.

plan8studios, 2016-09-21:
I'm having the same issue as @Thomas-Z on Sierra

mhennessie, 2016-09-23:
+1 

Never had an issue with this in 10.11.X but it is not working in 10.12.

schneems, 2016-11-02:
I was using the Strafe with a mac for about a year with no driver, and no problems. When I installed ckb then I'm no longer able to use CMD+click or shift+click. I'm currently using macOS Sierra.

For example CMD+click should open a new tab in chrome for me, but does not after installing ckb. Uninstalling ckb by turning it off and dragging it to the trash does not help, the problem still persists.

I set up the debug logging using this plist https://github.com/ccMSC/ckb/issues/70#issuecomment-107776269. I get this output

```
[I] Root controller ready at /var/run/ckb0
[I] Connecting Corsair STRAFE RGB Gaming Keyboar at /var/run/ckb1
[E] iterate_devices_hid (usb_mac.c:734): Failed to create device plugin: e00002c7
[I] Starting input thread for /var/run/ckb1
[E] loadrgb_kb (led_keyboard.c:211): Bad input header
[W] _start_dev (device.c:45): Unable to load hardware profile
[I] Setup finished for /var/run/ckb1
[W] iterate_devices_usb (usb_mac.c:629): Unable to seize USB handle, continuing anyway...
[W] add_usb (usb_mac.c:516): Tried to set up handle for device ckb1, but it was already set up. Skipping...
```

Is there anything else I can do to help debug the issue?  I'm going to try downgrading the firmware from http://forum.corsair.com/v3/showthread.php?t=138472 and see if that helps.

Thomas-Z, 2016-11-02:
Hello,

I think you just need to remove the plist mentionned here:

Uninstalling:
Drag ckb.app into the trash. If the system service file isn't cleaned up automatically, you can find it and remove it here: /Library/LaunchDaemons/com.ckb.daemon.plist.

schneems, 2016-11-02:
I tried downloading a firmware off of that page "1.30 FW For Strafe, K65, K70, K95" which gave me a `STRAFE_app130.bin` file. When I tried pointing ckb at this file it said it wasn't valid. Probably trying the wrong download.

Deleting the plist and the app solved the problem. Thanks! Unfortunately I have a corsair mouse as well that I would like to set the DPI of.

If there's other debug steps, that would hlep let me know.

mhennessie, 2016-11-02:
@schneems right now there is no real way to use a Corsair Keyboard in Sierra with CKB installed. I am not too familiar with Corsair mice but isn't there a DPI switch on the mouse that provides access to several saved setting? If so, set it up once and then remove CKB.