ccMSC/ckb #465: Corsair Gaming K70 LUX RGB "Keyboar" [sic] product ID 0x1b33 not recognized

tsmaster, 2016-09-16:
I have a K70 Lux RGB keyboard on my Ubuntu Linux machine, with the 2.05 firmware (I think), which I updated on a Windows machine.

lsub -d 1b1c tells me:
Bus 002 Device 010: ID 1b1c:1b33 Corsair 

I tried inserting the 0x1b33 product ID into usb.h, which leads to ckb-daemon recognizing the device, so far as connecting and identifying it as "Corsair Gaming K70 LUX RGB Keyboar" (no final 'd' on "keyboard").

ckb doesn't recognize any devices as connected, though.

I've tried newdev, which doesn't recognize my keyboard.

horen20wash, 2016-09-16:
The newdev branch has
 
#define P_K70_LUX       0x1b33
#define P_K70_LUX_STR   "1b33"

There is no need to update the usb entries.

tsmaster, 2016-09-16:
I must have pulled the newdev branch incorrectly - pulling it again today seems to work fine. Thank you.

closed 2016-09-16 by tsmaster

MrQuikenHD, 2016-11-18:
@tsmaster hey, how can you use the newdev branch? I really need to know :(