ccMSC/ckb #469: Upgraded to OSX Sierra, KeyRepeat is not working

mglinski, 2016-09-21:
I have a corsair k70 hooked up and running ckb v0.2.6 and after upgrading to ~OSX~ macOS Sierra the key repeat functionality is not working correctly.

I don't know where any relevant logs would be stored or how to enable debugging output to better test this, but I am a software dev and can follow instructions if you want to debug through me.

LMK how I can help find/fix this.

kchen13, 2016-09-21:
https://github.com/ccMSC/ckb/issues/452

He had issues with Sierra as well. I'm not sure if the info there can help but it seems in Sierra they rewrote the keyboard driver so maybe your issue coincides with the linked issue. Personally I'm waiting for this and some other things before I update.

horen20wash, 2016-09-21:
I have to machines running Sierra with ckb from the newdev branch. No problems at all.

kchen13, 2016-09-22:
shift-selecting multiple items and control-selecting multiple items

The above works for you? @horen20wash

jonathan-bird, 2016-09-22:
Doesn't work for me either. Subscribed

horen20wash, 2016-09-22:
shift-selecting multiple items does not work, key-repeat works.

mhennessie, 2016-09-22:
key-repeat is disabled by OS X so you can enter things like foreign characters by holding the key, similar to the function on an iOS device.

Shift+click or command+click is not working for me either.

mglinski, 2016-09-22:
@mhennessie Alt-Key Popups on my machine are disabled when holding down a key. I still get the key-repeat issue.

bnguyen86, 2016-09-22:
Command-click to select multiple items in Sublime and Command-click to open a new tab in Chrome also doesn't work for me either.  Just upgraded my MacBook Air to Sierra.

mhennessie, 2016-09-22:
@mglinski try entering this command in terminal:

defaults write -g ApplePressAndHoldEnabled -bool false

mglinski, 2016-09-22:
That is the option to disable the pop-over character selectors when holding down a key, which I stated is already active on my system. Key-Repeat is still broken with that setting active.

mhennessie, 2016-09-22:
@mglinski ok. I am not having issues with key repeat on my K65 RGB after upgrading to Sierra.

mglinski, 2016-09-22:
Ok. I am on my k70.

mhennessie, 2016-09-22:
@mglinski I previously had issues with key repeat when on El Capitan and that was what fixed it.

 - issue #327

mglinski, 2016-09-22:
Yea, everything was working correctly for me in El Capitan. I have uninstalled and reinstalled and played with every apple defaults setting I can find and nothing fixes it.

LadySleepsAlot, 2016-09-23:
Chiming in here, shift+click is not working for me either, worked perfectly before upgrading to Sierra.

shazron, 2016-09-23:
Ditto, shift/cmd click not working either. If I hold shift on the keyboard and move the up and down arrow keys, I can select, but not the previous non-macOS Sierra behaviour with shift-click

horen20wash, 2016-09-23:
Shift+mouse or Command+mouse doesn't work at all.

hjpx, 2016-09-25:
Shift Click and Command Click do not work

bwoodruff, 2016-09-25:
++ no keyboard modifier + mouse click actions work, including shift+click, ctrl+click, shift+right click (used in World of Warcraft), etc.

horen20wash, 2016-09-25:
That is correct, same here.

MasonMakram, 2016-09-27:
I'm so glad I'm not the only one. This software has been working so well for so long, I forgot that updating to Sierra might be an issue.

(Modifier) + Click doesn't work. Neither does key repeat.

InfiniteDevelopment, 2016-09-28:
I am having this same issue, and my ckb is saying "No devices connected".

mhennessie, 2016-09-28:
@InfiniteDevelopment when it says no devices connected, force quit the ckb-daemon. If that doesn't work then quit both CKB and the daemon.

horen20wash, 2016-09-28:
What is the keyboard you are using.

InfiniteDevelopment, 2016-09-28:
@mhennessie gonna try that now... @horen20wash k95 RGB

InfiniteDevelopment, 2016-09-28:
@mhennessie daemon wont force quit...

InfiniteDevelopment, 2016-09-28:
The force quit worked once I got it to go through. Command + Click && C,V,Q etc all works again...

Rog4242, 2016-10-03:
@InfiniteDevelopment  How did you manage to get the daemon to force quit? I've tried going through the activity monitor but it refuses to quit...

mhennessie, 2016-10-03:
@Rog4242 you might have to force quit it a few times. Also try force quitting CKB and then the daemon.

Rog4242, 2016-10-10:
@InfiniteDevelopment Thanks for the reply, I've tried everything to try to force it to quit, force quitting ckb then the daemon, vice versa, stopping the automatic start at login and restarting the computer, I can't find anything that works unfortunately.

Lutzmark, 2017-06-17:
OSX Sierra & Strafe RGB.Force quit CKB, unplug & replug the usb interface, then it works!