ccMSC/ckb #470: compile failed on macOS 10.12 Sierra

mrgaolei, 2016-09-22:
```
./os.h:72:13: error: typedef redefinition with different types ('int' vs
      'enum clockid_t')
typedef int clockid_t;
            ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/usr/include/time.h:171:3: note:
      previous definition is here
} clockid_t;
  ^
1 warning and 1 error generated.
make[2]: *** [debug/device.o] Error 1
make[1]: *** [debug] Error 2
make: *** [sub-src-ckb-daemon-make_first] Error 2
```

I'm using qt 5.7 and K65 RGB RAPIDFIRE

horen20wash, 2016-09-23:
Please comment out 

line 72
typedef int clockid_t  in src/ckb-daemon/os.h 

It will compile without problems.

mrgaolei, 2016-09-23:
@horen20wash when I do it, compile pass.
But in ckb.app show `No devices connected`
My Device id is: `0x1b39`
How do I debug what's wrong with me?

horen20wash, 2016-09-23:
Are you compiling the checkout from the newdev branch, right ? 

It has the right entry
#define P_K65_RFIRE     0x1b39
#define P_K65_RFIRE_STR "1b39"

You also followed the steps to install "service/launchd/com.ckb.daemon.plist" to "/Library/LaunchDaemons"

mrgaolei, 2016-09-23:
OK, I got it.

mrgaolei, 2016-09-23:
I forget to launchctl, thanks, It works fine.

closed 2016-09-23 by mrgaolei