ccMSC/ckb #471: ckb v0.2.6 No devices detected K65LUX

Midnightfender, 2016-09-23:
I am getting no devices connected when trying to use ckb. Using a K65LUX on macOS(sierra).

joonne, 2016-09-24:
Same for me using Ubuntu 16.04.

horen20wash, 2016-09-25:
You have to checkout and switch to the newdev branch and build ckb. The keyboard will work.

Midnightfender, 2016-09-28:
So Ive compiled the newdev branch and the exact same thing, no devices found, Im on the latest firmware as well.

Midnightfender, 2016-09-28:
Just for posterity sake I have also compiled and tried every available branch. None detect the device, reboots/disconnect-reconnects included.

Not sure if there was a firmware update. But I am running macOS Sierra as well.

horen20wash, 2016-09-28:
Every branch does not help, only newdev branch will support LUX keyboards. Did you verify that your device is is in the usb.h of the code you are compiling.

Midnightfender, 2016-09-28:
Yes, I actually got it working. I was forced to build it and fix a couple issues that popped up in the build debug. After complete build in Qt and demon relocation and linking everything seems functional. If I get a chance Ill be posting the prebuilt w/ demon install direcs.