ccMSC/ckb #473: Multiple issues with Sierra

BrekiTomasson, 2016-09-27:
Since upgrading to Sierra, the following things have stopped working (Corsair Gaming K95 RGB for reference):

* Binding to the windows key on the keyboard. This is where I normally bind "option", so it's a bit of an issue for me. Binding other things to the windows key doesn't work, binding option to either Ctrl or Alt works.
* Command-clicking things does not multi-select things.
* Holding down shift and clicking in another place to select everything in between (files or text) does not work.
* Holding down shift and pressing arrow buttons to select also does not work.

mhennessie, 2016-09-28:
@BrekiTomasson I am not experiencing an issue when binding the Windows key to Option. I have custom keys on my K65 RGB and I am positive it is working. I have bound Command to the Alt key and Option to the Windows key on both sides of the space bar.

BrekiTomasson, 2016-09-28:
@mhennessie I'm glad it works for you, but it's not working for me. :) Perhaps the K65 and K95 are different enough to make this not work as intended.

ljvasey, 2016-09-28:
Can confirm I have this issue as well, any combination of clicks + modifier keys does not work for me. macOS 10.12, K70 RGB.

bwoodruff, 2016-09-29:
I re-built ckb on Sierra thinking that might help but still no modifier key + click combinations work properly (e.x. shift+click)

smachi, 2016-10-01:
Same problem here with K70 RGB

MasonMakram, 2016-10-05:
Still no hope for this being fixed?

eduardbosch, 2016-10-06:
This is happening to me also with a _strafe rgb mechanical keyboard_. I hope this could be fixed, I don't like the idea of changing my keyboard.

Thanks for this great drivers! 😀

JSW88, 2016-10-06:
This is happening to me with a Daskeyboard 4 too. My Razer Blackwidow 2014 doesn't respond to anything either. Although I haven't tried plugging it straight into my laptop.

MasonMakram, 2016-10-11:
For the love of god, can someone track down ccMSC or find developer friends to fix this?

bluefurry, 2016-10-11:
I've been testing this out on my Mac, while command left click doesn't work in any of the browsers in OSX directly, while I'm in Fusion and a windows VM... command-click does work, almost seems like it's a OSX problem not a ckb problem.  I really hope a workaround is discovered, browsing is harder without that command-click

MasonMakram, 2016-10-11:
Well yeah. It's a Sierra issue. Someone said they rewrote how keyboards work in the OS. My dev friend suspects that they're gearing up for pressure-sensitive keyboards. Like, 3D Touch keyboards where you just press harder to capitalize a letter.

atarikguney, 2016-10-21:
I don't know how it worked before, but I purchased this keyboard after I already upgraded to Mac OS Sierra, but in my case, the keyboard is not recognized at all. Also, this project seems to be dead. It has been a while since the last commit. It seems we have nothing but waiting for an official support from the manufacturer itself, which seems not happening for anytime soon. At least, the most important functionality of the keyboard is running out of the box. People, however, can use a Windows machine and change the settings of the keyboard and saved them into the keyboard itself and come back to Mac.  And yes, the right Shift key is not working for me, which is another annoying problem.

MasonMakram, 2016-10-21:
So, this won't help anyone except the dumb people like me who are using CKB even though their keyboard isn't RGB.

When I first got my K70, the Option and Command keys were swapped. This drove me nuts, so I tried to remap them in the System Preferences. But for some reason, that wouldn't work. So I searched for a driver and found CKB. Installed it, used it to remap, and it worked great.

With the Sierra glitches, I gave up and put the K70 in my closet. But then I read someone asking, "Do you need CKB if your keyboard is only one color?" Someone said no. But I ignored it because I remembered the key remapping issue I had.

But that had me thinking...I deleted CKB and rebooted. The Shift + Click worked again. Then I opened the Modifier Keys settings in System Preferences. I just noticed THERE'S A "SELECT KEYBOARD" DROPDOWN, WHERE I HAD TO CHOOSE "CORSAIR K70R GAMING KEYBOARD"!!!!!!!!!!!!!

So my modifier keys are mapped properly. It works great! I'm so happy. I was just about to sell it.

Ferannor, 2016-10-21:
Did we have a possibility to completely disable remap function of CKB temporary until Apple fix this bug?

atarikguney, 2016-10-21:
@MasonMakram I don't think it helps smart people who know computers more than ordinary dudes. I am a developer myself, and it is not easy to fix someone's code. Also, the command line interface, I believe, is using the same underlying code that Graphical User Interface uses. Therefore, I don't think which one you use really matters here. Anyways, I hope some times someone who knows the code can address Mac OS Sierra issue soon. Otherwise, the manufacturer will have to provide something soon.

bluefurry, 2016-10-21:
There's totally a STRAFE keyboard in the pulldown when I remove cbt, but clicking didn't work, then I checked for the ckb-daemon... once I killed that the modifiers started working... don't really care about the lights anyhow... Thanks Mason!

bwoodruff, 2016-10-21:
@MasonMakram 

> So my modifier keys are mapped properly. It works great! I'm so happy. I was just about to sell it.

I've now switched to this method too, and uninstalled ckb.

But there is no way to use the keyboard lighting at all without ckb installed, right? I have a single color keyboard, and just want type lighting if possible, but I don't think it is without ckb?

Also I like to rebind the "windows lock" key as it has no real use on a Mac. I use it to lock my computer (running terminal command /System/Library/CoreServices/Menu\ Extras/user.menu/Contents/Resources/CGSession -suspend ). I don't think this is possible either without ckb?

I suppose I'm willing to give both of those up for working modifiers, but it'd be great to be able to have the best of both worlds.

mhennessie, 2016-10-21:
@bwoodruff there is no way to add type lighting without CKB that I am aware of. Also, I am not sure if OS X detects the windows lock key being pressed. You can create custom shortcuts using Automator Services and you would be able to make it execute that command but only if OS X detects that key input. I have done that to create an eject key for my optical drive using command+F15.

Guide for Automator Services: http://www.ostricher.com/2014/12/how-to-assign-keyboard-shortcuts-for-os-x-apps/

bwoodruff, 2016-10-21:
@mhennessie 

Thanks for the tip about automator (may come in handy for other things), but unfortunately macOS doesn't register the win lock key being pressed as a valid shortcut.

mhennessie, 2016-10-21:
@bwoodruff I suspected that. It is controlled by the hardware because it still locks the Windows key in OS X.

bwoodruff, 2016-10-23:
One other issue I've noticed with this is that the right Cmd key is not mapped correctly, only the left. Any fix for that?

mhennessie, 2016-10-23:
I'm not seeing that on my K65. I used the system prefs to swap command and option and the right and left command are working as they should. I do not have a right option though. On the bottom row my keys are ordered control, option, command, space, command, function, menu, control. I ideally I would have the menu key remapped to right option, used to be that way with CKB.

AlfredMan, 2016-11-15:
Same problem with my K95 with Sierra. Shift Click, Alt Click etc doesnt work on browser or with Photoshop etc, which is very important (shift click for adding point, alt click for removing point) Anyway. May need to remove ckb and just use the built in key remap from OS....

mire3212, 2016-11-15:
Strafe Keyboard and macOS Sierra also not working with modifier keys and mouse clicks.

AlfredMan, 2016-11-15:
Did that and yeah, that works. Miss the extra mapping CKB provided but I guess I just have to stay away from CKB until they updated it to Sierra

bwoodruff, 2016-11-15:
ckb has received no attention from the developer since June. It seems unlikely that we'll see any updates in the near future.

danielsdesk, 2016-11-16:
Do we actually know what changed in macOS Sierra to affect this? I'm not sure if I could help with efforts but I haven't seen anything that references what would need to be done to resolve this; does anyone know?  Any references to changes in APIs or whatever related?

wwvuillemot, 2016-11-24:
Unfortunately, I had to uninstall CKB on Mac OS Sierra to regain control of the modifier keys. Fortunately, System Settings for Keyboard allow you to compensate for the key flips for CMD and Option keys.

Miss the colors.  But missed more the basic functionality.

thecreativegeek, 2016-12-04:
Anyone got any fixes as of yet for OS Sierra ?

jeffreygnatek, 2016-12-27:
Removing CKB and Karabiner-Elements (just to be safe) then using the System Preferences > Keyboards > Modifier Keys to remap the alt/windows keys only and restarting fixed the issue of shift to select multiple etc. Not really a fix for CKB but at least made my keyboard usable.

jmc9981, 2017-01-03:
Anyone find a fix yet?

mattanger, 2017-01-04:
We've got a new fork that's under active development. Its not stable and
there is no support right now, but we're working on getting Sierra issues
resolved.

Feel free to try it out and help test, but don't expect any help until we
release a stable version.

https://github.com/mattanger/ckb-next

<https://github.com/ccMSC/ckb/issues/473#issuecomment-270127367>, or mute
the thread

danielsdesk, 2017-01-05:
Honestly I'm just glad something is happening.  Going to keep an eye on it and see if I can help in some way.

ryanaltvater, 2017-01-06:
I can confirm that Photoshop shortcuts aren't working on the K70 RGB either.

I'd rather not have to uninstall CKB, as I purchased the keyboard for it's lighting functionality. Hopefully we can figure out what changed with macOS Sierra and get a fix pushed. :) I'll take a look at ckb-next myself.

Handsomfreddy, 2017-01-16:
Hello, thanks in advance for your help and time in developing this awesome software.  Just reporting the same issue regarding holding down the shift key to select multiple pages in mac preview is not working.  

Also, holding down command key and clicking on a link in safari also not opening the link in a new tab or safari window.

SNARKYPOTATO, 2017-01-22:
Any luck with this yet?

Handsomfreddy, 2017-01-22:
Nope I believe Dev is working on it though. For now I installed color cue 2 from Corsair on windows and set a default static profile and saved it to the hardware and then changed the modifier keys on osX.  No effects but better then nothing.

SNARKYPOTATO, 2017-01-22:
@Handsomfreddy Okay, thanks. I've been reinstalling the keyboard using both USB connectors in system preferences. No effects but at least it works. Then I have to reinstall CKB when I'm done needing the modifiers.

krishunt, 2017-05-18:
There's a Mac binary of ckb-next on the github page now (Thanks, Matt!) Can anyone confirm that it works under macOS Sierra? I actually rolled back my system to El Capitan a while back so I could continue using my K70, and I don't want to install Sierra again until I know ckb-next going to be compatible.

krishunt, 2017-06-16:
ckb-next 0.2.8 only sort of works in Sierra. The Option key is not recognized in Photoshop.

mhennessie, 2017-06-16:
@krishunt you have to force quit the ckb-daemon on each boot. This is a known issue.

krishunt, 2017-06-16:
Ah, thanks. Is there hope of this being fixed in the future so you won't need to do that?

krishunt, 2017-06-16:
Does that mean opening Activity Monitor, and force quitting the process named "ckb"? Because that doesn't fix it for me.

mhennessie, 2017-06-16:
@krishunt It is just a known bug that has not been fixed yet.

thecreativegeek, 2017-06-16:
a new github has been setup for CKB by a new dev team and these issue have been fixed with the new version you can view it here https://github.com/mattanger/ckb-next

krishunt, 2017-06-16:
CAPS LOCK IS STUCK NOW... :/  IS THIS ALSO A KNOWN BUG?

thecreativegeek, 2017-06-16:
Try using the new version I linked to kris hardly any bugs on the new GitHub as it's been completely redeveloped 👍

krishunt, 2017-06-16:
Isn't that just a link to the same 0.2.8 version I'm already using? The only link that exists for a Mac binary, in fact?

mhennessie, 2017-06-17:
@krishunt v0.2.8 of ckb-next is the only one for Sierra. The caps lock is also a known bug and there is a mention of changing it in the settings somewhere on here. I'm on my phone so I'm not able to find the exact mention of the setting, it might be setting it to RGB Only though.

krishunt, 2017-06-20:
I'm going to have to get a new keyboard, as much as I hate to see the K70 RGB go. Are all of these issues specific to all RGB keyboards, or all Corsair keyboards, or just Corsair RGB keyboards? Would a Corsair Vengeance K70 (red backlight) work fine in Sierra?

mhennessie, 2017-06-20:
@krishunt what issues are you having? I only have one with my K65-RGB and it is fixable by force quitting the ckb-daemon in activity monitor. You also don't have to use CKB at all and the keyboard will work fine but you lose the lighting effects. You can use Windows to set a static profile footer the lighting and save it to the device.

krishunt, 2017-06-20:
Well you know, I'd always heard that you could use the keyboard without ckb running, but whenever I tried, I found it was impossible to swap the modifier keys with **System Preferences > Keyboard > Modifier Keys**. I tried it again just now and still had that same problem. But then I checked Activity Monitor, and found that ckb-daemon was still running (even though ckb was not), and after force quitting it, I'm able to swap the modifier keys with no problems. And my backlighting is solid white, so that works out just fine. So maybe things are just fine as they are. Thanks for the tip.

mhennessie, 2017-06-20:
You need to make sure you remove the daemon file. Trashing the ckb-daemon app isn't enough to remove it all.

krishunt, 2017-06-20:
I didn't realize the daemon was an actual file. Where's it located?

mhennessie, 2017-06-20:
/System/Library/LaunchDaemons

krishunt, 2017-06-20:
I don't see a ckb file in that directory, but I do see one in /Library/LaunchDaemons.

mhennessie, 2017-06-20:
Delete that file. I couldnt remember if it was in the user library or the system library.

daclink, 2017-09-11:
It seems like the latest version of https://github.com/mattanger/ckb-next has fixed this issue... at least it seems to be working for me. =-)

closed 2021-07-10 by BrekiTomasson