ccMSC/ckb #478: Is this project still being maintained?

owenzhao, 2016-09-29:
It seems that there has been no updates since June 30, 2015.

bwoodruff, 2016-09-29:
It definitely does seem progress has slowed.

@ccMSC is there anything we can do to help move the project forward? I'm not a developer myself so submitting PRs is likely out of the question.

horen20wash, 2016-10-01:
It is my impression as well. It either slowed down to an halt or is abandoned.

fleischie, 2016-10-01:
Please consider donating to the maintainer. The more you donate, the more she/he is willing to take care of the various problems caused by OS X. :wink:

horen20wash, 2016-10-01:
I donated more than once. That's not the point. 
If the maintainer does not have the time for any reason, someone else may take over the project and support. A clarification would help.

BrekiTomasson, 2016-10-04:
I think either way, the inactivity since Sierra's beta is quite telling. I think somebody should just grab the code and run with it, making their own thing from here on in at this point. Anything is better than not having any code commit or comment by @ccMSC since what; late June?

supahkutdiamnds, 2016-10-05:
yeah, it doesn't work for my M65 RGB (optical). I bought this mouse off the retail shelves likes last week and decided to get ubuntu going again....doesn't work for it at all.

I'm just a java guy. Anyone know the extent of C++ required to maintain or even tweak this project?

frickler24, 2016-10-13:
@fleischie You get the point: If we can find someone with OSX devel know how, we have a chance to bring this project on the road again. I have C / C++ development know how, but I do not own an apple or hackintosh installation (using linux for my  development). 
Next problem is the rising number of new hardware in the last months. Do we have a chance getting  an overview which developer / tester has what HW / OS-combination?

frickler24, 2016-10-15:
No reaction?
OK, I try to start with some Information:

System: Linux Mint 17.3 kde, 18 xfce; maybe other linuxe if necessary
Corsair hardware available:  
  K95 RGB
  M65 RGB

skills relevant for maintaining ckb:
 good: 
    C, C++ (I did the develop for G-macro keys), 
    scripting
    testing
    code review
 little: 
    qt / GUI (the ckb UI for key-settings drive me crazy), 
    system drivers or user space drivers,
    git
 none: 
    Mac
    network monitoring for new hardware protocols
    qmake

time available for doing something with ckb:
  about 3-4 hours a week, mostly at weekend

TwoD, 2016-10-15:
Perhaps make a wiki page where one can post their interest, skills and hardware?
I fear this thread will be quite long and details may be lost amongst all comments if it grows.

Anyway; I don't think I have time to actually dig in and write code for this (would be fun though), but I do have some C/C++ experience though I mostly work in PHP/JS now. Running Linux pretty much everywhere and I have a K95 RGB only at work so won't be able to continously test new changes but would be interested in testing extended macro support. At work we also have 3 other K95 RGB's connected to Linux, Windows and MacOS boxes, all but the Windows one run ckb though I'm not sure if they would be interested in being testbeds.

petrklus, 2016-10-24:
+1 @ccMSC - care to weigh in?