ccMSC/ckb #482: K65 LUX FREEZING ON CAPS LOCK

Midnightfender, 2016-10-01:
TO START OFF WITH SORRY FOR THE CAPS: FOR SOME REASON MY LUX IS FREEZING WHEN EVER I HAVE CAPS LOCK NOT ON. EVEN WHEN DISABLED IF I DO NOT HAVE CAPS ON (WHICH ISNT LIGHTING UP ON THE BOARD) IT WILL FREEZE THE ENTIRE SYSTEM EVERY 10-20 SECS. ANY HELP WOULD BE USEFUL.

smachi, 2016-10-03:
Same problem here after updating to MacOS Sierra

mhennessie, 2016-10-04:
Also getting this on a K65 RGB non-LUX in macOS Sierra.

deejayfrankcesco, 2016-10-05:
same here with macOS sierra and K95

jordan4302, 2016-10-07:
Same issue here with Sierra and a K70 LUX. As soon as I press caps lock my whole mac starts freezing every few seconds, but it stops if I kill ckb. Something else interesting to note is that after I do that ckb no longer recognizes my keyboard until I restart.

mhennessie, 2016-10-07:
@jordan4302 kill the ckb-daemon too.

I have had success by unbinding the caps lock key. Obviously I loose the caps lock functionality but it keeps macOS from freezing if I accidentally hit that key.

deejayfrankcesco, 2016-10-07:
Try to select slower polling frequency rate with BIOS button. In my case with 2ms or 4ms caps lock works well without freezing.

jordan4302, 2016-10-07:
@deejayfrankcesco Still freezes for me even with lower polling rates.
@mhennessie Thanks for the idea of unbinding the key. Didn't think of that and I had just removed the keycap.

In case this is also related, is anyone else noticing the lighting on the keyboard sometimes being choppy/laggy/not smooth?

smachi, 2016-10-10:
I'm not sure it's related but since I changed the Caps Lock lights from Normal to RGB indicator and restarted my computer, this issue stopped from happening me

mhennessie, 2016-10-10:
@smachi that's interesting and I will try it this evening when I get home. I did notice that when I was having the issue the caps lock indicator led on the top right was stuck on wether caps lock was on or not.

jordan4302, 2016-10-10:
I did some testing after what @smachi said and its seems to be that its when ckb tries to control the caps lock LED that things start going screwy. I also noticed that the LED still indicates caps lock status regardless of ckb's settings.

mhennessie, 2016-10-11:
@jordan4302 yea, setting caps lock to RGB solves the freezing problem but the led on the upper right still lights up regardless.

jachicao, 2016-12-13:
can confirm on my Strafe keyboard (lol that S freezed it)

tatokis, 2016-12-13:
Does it happen if you type uppercase letters with the shift key, instead of caps lock?

jachicao, 2016-12-13:
Both actually.

contreras2004, 2017-05-04:
Anyone could resolve this?

mhennessie, 2017-05-04:
@conteras2004 I have just been using my K65 RGB without the software. Besides not having controls over the lighting it works perfect.

tatokis, 2017-05-04:
https://github.com/mattanger/ckb-next/issues/23#issuecomment-275820468

contreras2004, 2017-05-04:
@mhennessie I found a solution in another thread: https://github.com/mattanger/ckb-next/issues/23