ccMSC/ckb #486: Lost SW Profiles

deejayfrankcesco, 2016-10-05:


deejayfrankcesco, 2016-10-05:
I use ckb with K95 on macOS sierra. All works perfectly but at last reboot ckb lost all saved Profiles/Layout ecc. It seems that the keyboard has stored some lights settings because when I start the macbook there are the correct light colours on the key but when ckb starts all keys become white and all "factory reset" settings are loaded. I lost a lot of time to set all the terminal command associated whit keys but now It seems that I cannot recovery anything.

MasonMakram, 2016-10-07:
Does your Modifier + Click work?

deejayfrankcesco, 2016-10-07:
yes, perfectly! 
At now, I remapped all keyboard and It works. I only hope that It doesn't lost data in future reboot, unplugging etc

deejayfrankcesco, 2016-10-07:
I'm sorry sir. Modifier keys works well until now ! Now cmd keys does not works!

ggoyanes, 2016-10-09:
Same issue here. After a reboot my profiles where gone. Before daemn starts I can see the correct configuration of colours. I'm in a Kubuntu Linux. and a K70 RGB keyboard

deejayfrankcesco, 2016-10-11:
Same again, I formatted system for other issues and then I retried to install ckb. All works ok, Modifier keys works well too. I was very careful to do not unplug the usb cable of keyboard or do something  unusual but when I reboot last time and I lost configuration again. Modifier keys still works

joinscome, 2016-10-30:
Same issue here.

jkyeo, 2016-11-07:
same issue here 
Mac Os X 10.11.6
Cannot restore even reinstall ckb

mupman, 2017-04-01:
same issue, macOS Sierra 10.12.4

this is happening on my Corsair Sabre RGB mouse

uninstalling CKB with AppDelete and then reinstalling will restore the default profiles, but they'll be gone again after a reboot or two

to try and counter it wiping out all the profiles, i tried duplicating the 3 original profiles.  however i have just rebooted and both they and the original profiles are all gone.  the 'Rainbow' profile remains, however it just shows 1 LED set to green and the rest white - the original rainbow effect is gone.