ccMSC/ckb #488: K70 LUX RGB build?

AwpData, 2016-10-09:
Can someone build me a build that supports the K70 LUX RGB for mac please? I'm no developer and I am so lost on how to build it. My main problem though is that it says no devices connected. I have tried everything on all the articles specifically about this keyboard and they don't work. Thanks for the help.

slugger7, 2016-10-10:
Im on Linux also have nothing. At first it didnt work at all unless i switched it to BIOS mode but did a firmware upgrade from windows now it works but CKB software does not pick it up as a device.

slugger7, 2016-10-10:
I pulled the newdev branch and installed that and it worked for me. There is only two tiny issues but it doesnt bother me though.

mhennessie, 2016-10-10:
Follow the instructions here from 4t0m1k: https://github.com/ccMSC/ckb/issues/298#issuecomment-216057774

drohm, 2016-10-11:
I've downloaded the 'newdev' branch and did ./quickinstall, but it still doesn't see my K70 LUX RGB (No devices connected).

I booted into windows and installed the latest firmware: 2.05.  What firmware version are you running, @slugger7 ?

slugger7, 2016-10-11:
I am running the exact same firmware on mine (2.05) But my keyboard is working without a hitch.

drohm, 2016-10-11:
Thanks @slugger7. I downloaded the [newbranch.zip](https://github.com/ccMSC/ckb/archive/newdev.zip) file from [this issue,](https://github.com/ccMSC/ckb/issues/435#issuecomment-239241399) and it works great.  I wonder why pulling directly from the branch and installing from that didn't work.

slugger7, 2016-10-11:
Glad i could help.

AwpData, 2016-10-11:
@slugger7 @drohm  My keyboard is also at version 2.05 but I can't seem to quickinstall from the newbranch.  I download Qt5 and Xcode but it says this:

Please make sure you have Qt5 installed.
Re-run this program with QMAKE=/path/to/qmake to select qmake manually.

What is Qmake and how do I access it?

drohm, 2016-10-12:
@AwpData Try downloading the newbranch.zip file I linked in my previous comment and install that.  That version worked for me.

AwpData, 2016-10-12:
@drohm sorry, what do I install? The quickinstall thingy?

drohm, 2016-10-12:
Yes, untar it and launch ./quickinstall.

AwpData, 2016-10-12:
@drohm ok I will try and tell you the results

AwpData, 2016-10-12:
@drohm I got this as a result:

readlink: illegal option -- f
usage: readlink [-n] [file ...]
Preparing build files...
No suitable qmake found.
Please make sure you have Qt5 installed.
Re-run this program with QMAKE=/path/to/qmake to select qmake manually.

drohm, 2016-10-12:
Looks like you need to install the prerequisites in order to build.  Look in the README.md under the 'Preparation' section.  I'm not sure which distribution you're using, but the README.md lists the libraries needed in order to build for Ubuntu, Fedora, Arch, and others.

AwpData, 2016-10-12:
@drohm Ok, stay tuned...

AwpData, 2016-10-12:
@drohm I don't see a preparation section for MAC. I think you are talking about the Linux section?

AwpData, 2016-10-12:
@drohm Anyway I have already followed all the requirements for the 'building from source'. I just can't get the ./quickinstall to run it's so frustrating!

drohm, 2016-10-12:
Apologies about that, I'm using linux.  I think you'll need someone with more MAC experience to assist you further.

AwpData, 2016-10-12:
@drohm Thanks for the help anyway. Can you think of any possible ways to get this installed on Mac or fix this qmake issue?

drohm, 2016-10-12:
Did you do the following:

> #### Building from source:
> 
> Install the latest version of Xcode from the App Store. Open Xcode, accept the license agreement, and wait for it to install any additional components (if necessary). When you see the "Welcome to Xcode" screen, the setup is finished and you can close the app. Then install Qt5 from here: http://www.qt.io/download-open-source/

In particular, did you download Qt5 and install it?

AwpData, 2016-10-12:
What should the app be called? 

I have Qt creator. Is that the app??

drohm, 2016-10-12:
I don't have a Mac so I'm not sure how much help I can give you here.  Apologies.

AwpData, 2016-10-12:
@drohm That's fine. Can you see if you can get some OSX users onto this thread to help me? That would be great.

slugger7, 2016-10-12:
@AwpData I am also running Linux soo I cannot really help you there. I think you need to get the computer administrator (an account with sudo privileges) to install all of the software in order for you to be able to install all the software needed to build the application and install it. (Not sure but that might be the problem)

AwpData, 2016-10-13:
@slugger7 Do you know what qmake is?

slugger7, 2016-10-13:
@AwpData https://en.wikipedia.org/wiki/Qmake

mhennessie, 2016-10-13:
Follow the instructions at the link I posted earlier. That is for Mac.

AwpData, 2016-10-14:
@mhennessie Ok will try. I will tell you the outcome

AwpData, 2016-10-14:
@mhennessie So I was able to put the file into LaunchDaemons but as I try to enter the commands I get this: (Username) is not in the sudoers file.

AwpData, 2016-10-20:
@slugger7 @mhennessie @drohm @ccMSC Hi everyone, just like to show you that this issue is still open and my comment above is the problem I am having right now. Please help! (If you can't understand what I am trying to say, look at the above comments to understand better) :P

mhennessie, 2016-10-20:
@AwpData I didn't mean for you to follow those instructions but rather the part at the bottom on that post:

> Building by yourself is very easy if you have Xcode and Qt5, in Terminal, go to the ckb-fw2kb folder (from .zip that you download on github) and launch ./quickinstall
> 
> If your Qt installation is not the default one, you have to declare the QMAKE env var. My command for example : QMAKE=/Developer/Qt/5.5/clang_64/bin/qmake ./quickinstall
> 
> Then, at the end of the install, it will propose you to install it, press 'y' 3 times.

You need to download and install QT5 and any prerequisites and then just run the ./quickinstall comment.

AwpData, 2016-10-20:
@mhennessie I do have QT5 installed. Not sure why the quickinstall won't install.

closed 2018-12-11 by AwpData