ccMSC/ckb #490: K70 LUX RGB + newdev + sierra = broken cmd keys

rkitover, 2016-10-14:
I had to make a minor change for newdev to build on Sierra, don't know if this is correct:

```patch
diff --git a/src/ckb-daemon/os.h b/src/ckb-daemon/os.h
index 0e26d21..4eb5e45 100644
--- a/src/ckb-daemon/os.h
+++ b/src/ckb-daemon/os.h
@@ -69,7 +69,7 @@ void *memrchr(const void *s, int c, size_t n);

 #define CLOCK_MONOTONIC 1
 #define TIMER_ABSTIME   1
-typedef int clockid_t;
+//typedef int clockid_t;
 int clock_gettime(clockid_t clk_id, struct timespec *tp);
 int clock_nanosleep(clockid_t clock_id, int flags, const struct timespec *rqtp, struct timespec *rmtp);

```

It seems there is a clockid_t type in the headers in Sierra.

So this builds, and I was able to use one of the animations (the others seem broken.) But now I have a problem, it seems my cmd keys aren't working the way they are supposed to. E.g. in iterm2 I need to hold cmd to click on a link, and I can't, this is what happens:

https://gitlab.com/gnachman/iterm2/issues/5188#note_16113297

I did not notice this because everywhere else cmd works just fine, all system shortcuts etc. work correctly.

brettnem, 2016-10-17:
I'm also having this issue. Seems that any kind of command-click or shift-click (like to multi-select) just doesn't work.

mhennessie, 2016-10-18:
This is a known issue in Sierra

rkitover, 2016-10-18:
Would you advise uninstalling ckb for the time being?

mhennessie, 2016-10-18:
@rkitover personally, I uninstalled it and the keyboard is working better but lacks lighting effects. I booted into Windows and setup a static lighting profile that I would use for OS X and saved that to the hardware so it shows up that way in OS X. The media keys still work and command/shift/control + mouse click work fine. I remapped the option and command keys using the default OS X system preferences.

atarikguney, 2016-10-21:
@rkitover I think you can change the variable name to something else instead of commenting it out. In fact, I will try that and will post the results here.

atarikguney, 2016-10-21:
I have changed the variable name `clock_t` to `clock_new` and was not helpful. Now we can focus on somewhere else.

```
#define CLOCK_MONOTONIC 1
#define TIMER_ABSTIME   1
typedef int clock_new;
int clock_gettime(clock_new clk_id, struct timespec *tp);
int clock_nanosleep(clock_new clock_id, int flags, const struct timespec *rqtp, struct timespec *rmtp);
```

bluefurry, 2016-10-21:
Uninstalling works, but you also need to kill the daemon by resetting your mac or doing it in a shell with "killall ckb-daemon"
We lose the awesome lighting, but at least the keyboards are usable again :) Thanks all for the help!

rkitover, 2016-10-23:
I tried running the windows software with the kbd connected to my windows VM, it is atrocious, I couldn't figure out how to do anything in it, and it made my VM BSOD a few times and spin the CPU. In particular how to save any kind of lighting profile to the kbd. I was trying version 2, I have heard some people say you should use the version 1 which is still available on their download site. I'll try that at some point.

mhennessie, 2016-10-23:
I run Corsair CUE on a native boot of Windows 10 and don't have any issue. I think I'm running version 2.4 something.