ccMSC/ckb #495: CKB periodically starts working until reinstallation

Kristaphonie, 2016-10-18:
There might be another issue for this already, so apologies if that's the case. 

But I'm now to a point with CKB that after I install it, a couple of weeks afterwards (or so) it will stop recognizing my keyboard and will go to the default white and red color scheme. My Mac still recognizes it, and I've set up the keyboard system settings to map the keys to how I need it to survive on mac, but no color control. 

After this happens I will go onto the newdev branch and rerun the quickinstall script. This will every time fix the issue (at least the past 3 times I've done it).

I don't know if the OS is writing over certain files or what, but just thought I should bring it up.

It's not a huge debilitating thing for me, since I'm able to still use the keyboard, my themes just help calm me during my work day. :)

mhennessie, 2016-10-18:
@Kristaphonie just force quit the ckbdaemon using Activity Monitor. It may take a few tries to force quit it. I would recommend changing one light from the stock WASD and arrow keys being white because it will give you a visual cue that CKB isn't working properly. After you force quit the CKB then your lighting and key profile should load.

Kristaphonie, 2016-10-19:
This morning, after fixing it yesterday, when I plugged in my keyboard after getting into work it stopped working. I rebuilt newDev and it started working again.

mhennessie, 2016-10-19:
@Kristaphonie I am pretty sure you could just force quit the ckb-daemon to get it working again without having to reinstall it over and over.

Kristaphonie, 2016-10-19:
Quitting won't do any good. It's not the application running is the problem, it's the application no longer recognizing the keyboard, which is probably happening because the OS is writing over the config files for CKB (I assume, but may not be the case).

I closed and reopened it a few times, which didn't gain me anything. Every time this has happened it has only started working again after I reinstalled. A couple of reinstalls ago, I spent quite the amount of time trying to troubleshoot this, and thus far reinstalling has been the only fix. :/

mhennessie, 2016-10-19:
@Kristaphonie you have to force quit the ckb-daemon specifically, not the CKB app. Quitting the app won't do anything because the cab-daemon is still running in the background. If you are running OS X then use Activity Monitor to find and force quit the daemon.

Kristaphonie, 2016-10-26:
So now it's not even working whether I reinstall it or force quit it via activity monitor. It's just straight up not recognizing the keyboard now. :(