ccMSC/ckb #496: CKB not working with K70 Lux RGB, iMac & OS-X 10.12

juniorwoo, 2016-10-19:
Have tried multiple installs, reboots and all possible combinations (single and double) of USB ports.

I'm a complete dope when it comes to command line editing and it scares the hell out of me, but it there an easy work-around?

mhennessie, 2016-10-19:
@juniorwoo there are lots of issues with CKB and macOS Sierra. The developer hasn't posted/commented or updated anything since June.

Currently I am running my K65 RGB without CKB and it is working 95%. There are obviously no lighting effects and you can't rebind a few keys but it is working much better without CKB. You have to completely uninstall CKB and the daemon though. For lighting, I booted into Windows and setup a profile I liked for OS X and then saved that to the hardware. When you boot into OS X, the keyboard will be lit with that profile.

atarikguney, 2016-10-21:
@mhennessie I think that's the only way possible for Mac Sierra users. At least, they have the hardware saving option. @juniorwoo, the manual build is not working either. So there is no need to use command-line, except for the fact that there is a command-line interface for configuring the keyboard, and I bet it would not work either since they probably use the same underlying code that gives live to both GUI and command-line interfaces. It is so sad that the manufacturer does not have a driver for Mac users for an awesome keyboard like K65. I have purchased one today, and I have been enjoying it quite a lot, even though it is louder than mushy keyboards.

MrQuikenHD, 2016-11-15:
@mhennessie I did that aswell. Now, I can only do this with the static profile :/ Is this a problem or is it normal?

mhennessie, 2016-11-16:
@MrQuikenHD yes you will be stuck with a static profile without CKB. It is really the only way the keyboard is usable in OS X though.

MrQuikenHD, 2016-11-16:
@mhennessie and do you know anyway to get the K70 Lux RGB working with ckb on OS X 10.9.5?