ccMSC/ckb #497: K95 does not work in Sierra

deejayfrankcesco, 2016-10-21:
Unfortunately I experimented several instabilities with macOS Sierra. 
At first installation all was ok but:

- At next reboot I can't insert login password because system freezes periodically
- Similar freeze problems with Caps Lock
- Sometimes It causes kernel panic
- Sometimes the layout/settings are lost after reboot

I hope that ccMSC can resolve all this issues. I'm sure He will do ! :)

PS. I suggest a sort of layout import/export function. It could be very useful for computer migration and backup.

mhennessie, 2016-10-22:
Try removing CKB altogether. All of your issues have been report before and it is not limited to your model keyboard.

clodal, 2016-11-12:
Just got the K95 RGB. Installed ckb on macOs Sierra. Lights worked and macro keys detected. But modifier keys (Cmd/Shift/Fn/Ctrl/Opt) + click don't work. After hours of trying, decided to uninstall ckb as suggested by @mhennessie and the modifier keys + click work again. Downside is that the macro keys and lights are not working. Hoping for a patch someday. Good luck everyone!