ccMSC/ckb #498: ckb K70 LUX Sierra problem

DeviousR, 2016-10-22:
My ckb won't work. It says that Xcode isn't running properly and says I have to do something with the license agreement. PLEASE HELP

mhennessie, 2016-10-22:
CKB causes lots of issues in Sierra. I suggest not using it.

atarikguney, 2016-10-23:
@DeviousR Don't waste your time. There is a solution to that problem, but it does not help eventually. If you want to change something with your keyboard, make the change via a Windows computer, and persist your changes inside the keyboard by saving them into the keyboard itself, which is the default option for most of the possible operations. Then, continue on Mac.

Ferannor, 2016-10-23:
Agree, I compiled it on Sierra, add a fix around on xcode 8 but compatibility is really tricky. 
Apple has change some API and there to many problems to use it for now. So forget it at this time.