ccMSC/ckb #501: Mac - K65 LUX w/ firmware 2.05 not detected...

soleil4, 2016-10-25:
Hello,

I have compiled ckp.app from the newdev branch, while having the latest Corsair firmware (2.05) and am still having trouble getting ckb to recognize the K65 LUX RGB. Can anyone with knowledge on this issue give me some advice on how to fix this? (I have a feeling I compiled the app incorrectly as I have both the latest firmware and can use the keyboard normally to type)

Thanks

chrisRidgers, 2016-10-28:
I did the same recently: was still a slight struggle.

If you did what I did and attempted to compile after trying the main branch make sure that you've completely replaced files from the original install.  Remove them, and then copy your compiled dev branch files into place.

And follow the instructions for configuring the service/ application interface on the build page.

(Even after all that, with the software running, I found that initially I had to unplug the keyboard and replug to get any kind of feed back, since then its been working fine from boot, with the exception of my F1..F12 keys in OS X).

chrisRidgers, 2016-10-28:
https://github.com/ccMSC/ckb/issues/489

My open Issue regarding the FKeys