ccMSC/ckb #504: Ubuntu: M65 RGB mouse has usb ID 1b1c:1b05, changing it in usb.h locks the device

samvde, 2016-10-28:
I have a new M65 RGB mouse that works but did not show up on ckb.

I quickly checked the src/ckb-daemon/usb.h file and it seems the usb ID differs from what I have. Mine has 1b1c:1b05, while usb.h defines 1b1c:1b12 for the device. After adapting this file as well as src/ckb/fwupgradedialog.cpp (and running quickinstal again) the device does show up in ckb, but the mouse disconnects from the system unless I stop the daemon. So that's actually a worse outcome unfortunately.

I tried adding the usb quirk option to the kernel boot parameters, to no effect.

I have tested this on Ubuntu 16.10 with kernel 4.8.0-26-generic. I'll probably do a 16.04 test this weekend. I can help testing but I don't know how to code properly.

frickler24, 2016-10-29:
Your M65 is a [Vengeance mouse](http://www.linux-hardware-guide.com/uk/2014-08-16-corsair-vengeance-m65-usb-gaming-mouse-8200-dpi-laser). I cannot test it, because I just own the older one.
Question: Why do you stop the daemon? The daemon is used for running the mouse. If you stop it, you stop the mouse activity and color. And it's a daemon, so its job is to run everytime :-)

samvde, 2016-10-29:
Hi,

Yes, I did some research and found this out just now. I assume a lot of people mentioning this issue have the same ("different") device as me.

I had to stop the daemon once I added the USB identifier to usb.h, but now that makes sense.

Still available to help and test though!

broedli, 2016-12-22:
I hope you don't mind me chiming in on this issue.

I'm having similar issues with my M65 Vengeance (Full [lsusb -vv](https://gist.github.com/broedli/a89ffbaca2399b9c842a0b4bf5d39631)):
```
$ lsusb | grep Corsair
Bus 001 Device 006: ID 1b1c:1b05 Corsair
```
I've changed the Product ID in [usb.h](https://github.com/ccMSC/ckb/blob/master/src/ckb-daemon/usb.h#L33-L34) and [fwupgradedialog.cpp](https://github.com/ccMSC/ckb/blob/master/src/ckb/fwupgradedialog.cpp#L22) from `1b12` to `1b05` which leads to a detection inside of ckb but leaves the mouse cursor without any response:
```
$ sudo ckb-daemon
    ckb: Corsair RGB driver beta-v0.2.6
[I] Root controller ready at /dev/input/ckb0
[I] Connecting Corsair M65 Gaming Mouse at /dev/input/ckb1
[I] Starting input thread for /dev/input/ckb1
[E] os_usbsend (via firmware.c:15): Broken pipe
[W] _start_dev (device.c:24): Unable to load firmware version/poll rate
[I] Setup finished for /dev/input/ckb1
```
```
$ journalctl -f
Dec 22 16:08:46 workstation.domain.tld sudo[13262]:     $USER : TTY=pts/2 ; PWD=/home/$USER ; USER=root ; COMMAND=/usr/bin/ckb-daemon
Dec 22 16:08:46 workstation.domain.tld sudo[13262]: pam_unix(sudo:session): session opened for user root by (uid=0)
Dec 22 16:08:46 workstation.domain.tld kernel: input: ckb1: Corsair M65 Gaming Mouse as /devices/virtual/input/input35
Dec 22 16:08:46 workstation.domain.tld kernel: input: ckb1: Corsair M65 Gaming Mouse as /devices/virtual/input/input36
```

When I start the ckb UI I get the following inside `journalctl`:
```
$ sudo ckb-daemon
...
[E] os_usbsend (via device_mouse.c:30): No such file or directory
[I] Attempting reset...
[E] os_usbsend (via device_mouse.c:30): No such file or directory
(repeats last msg constantly)
```
```
$ journalctl -f
...
Dec 22 16:08:46 workstation.domain.tld ckb[2048]: Starting  "/usr/bin/ckb-animations/ckb-wave"
Dec 22 16:08:46 workstation.domain.tld kernel: usb 1-10: reset full-speed USB device number 6 using xhci_hcd
(repeats last msg constantly)
Dec 22 16:08:59 workstation.domain.tld kernel: usb 1-10: usbfs: process 13276 (ckb-daemon) did not claim interface 0 before use
(this one is pretty random it seems but it pops up once in a while)
```
This result doesn't seem to be influenced by the Kernel Parameter but here is the log of a boot with the parameter set (note that I've set the correct product ID inside the parameter)
```
Dec 22 16:02:51 workstation.domain.tld kernel: Command line: BOOT_IMAGE=/vmlinuz-linux root=UUID=486da7b0-d26e-48f3-bcb6-5daac7c49bed rw rootflags=subvol=__active/rootvol intel_iommu=on quiet usbhid.quirks=0x1B1C:0x1B05:0x20000408
Dec 22 16:02:51 workstation.domain.tld kernel: Kernel command line: BOOT_IMAGE=/vmlinuz-linux root=UUID=486da7b0-d26e-48f3-bcb6-5daac7c49bed rw rootflags=subvol=__active/rootvol intel_iommu=on quiet usbhid.quirks=0x1B1C:0x1B05:0x20000408
Dec 22 16:02:51 workstation.domain.tld kernel: input: Corsair Corsair M65 Gaming Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10:1.0/0003:1B1C:1B05.0004/input/input4
Dec 22 16:02:51 workstation.domain.tld kernel: hid-generic 0003:1B1C:1B05.0004: input,hidraw3: USB HID v1.11 Mouse [Corsair Corsair M65 Gaming Mouse] on usb-0000:00:14.0-10/input0
Dec 22 16:02:51 workstation.domain.tld kernel: input: Corsair Corsair M65 Gaming Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10:1.1/0003:1B1C:1B05.0005/input/input5
Dec 22 16:02:51 workstation.domain.tld kernel: hid-generic 0003:1B1C:1B05.0005: input,hidraw4: USB HID v1.11 Keyboard [Corsair Corsair M65 Gaming Mouse] on usb-0000:00:14.0-10/input1
Dec 22 16:02:51 workstation.domain.tld kernel: input: Corsair Corsair M65 Gaming Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10:1.2/0003:1B1C:1B05.0006/input/input6
Dec 22 16:02:51 workstation.domain.tld kernel: hid-generic 0003:1B1C:1B05.0006: input,hidraw5: USB HID v1.11 Device [Corsair Corsair M65 Gaming Mouse] on usb-0000:00:14.0-10/input2
```
The result is the same if I enable `ckb-daemon.service` and reboot, start the service manually or just execute the binary. The mouse is always unresponsive (not reacting at all) after `ckb-daemon` is started and only becomes responsive after terminating the daemon
Obviously there seems to be a problem with the `device_mouse.c` file being missing but I am lost on how to fix this and what location it's being searched for.

If you've fixed this issue, could you elaborate on how you did it?