ccMSC/ckb #506: K70 Lux

0mni, 2016-10-30:
```
Oct 30 00:14:42 desktop kernel: usb 3-2: unable to read config index 0 descriptor/all
Oct 30 00:14:42 desktop kernel: usb 3-2: can't read configurations, error -110
Oct 30 00:14:42 desktop kernel: usb 3-2: new full-speed USB device number 4 using xhci_hcd
Oct 30 00:14:42 desktop kernel: input: Corsair Corsair Gaming K70 LUX Keyboard  as /devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2:1.0/0003:1B1C:1B36.0002/input/input20
Oct 30 00:14:42 desktop kernel: hid-generic 0003:1B1C:1B36.0002: input,hidraw1: USB HID v1.11 Keyboard [Corsair Corsair Gaming K70 LUX Keyboard ] on usb-0000:00:14.0-2/input0
Oct 30 00:14:42 desktop kernel: input: Corsair Corsair Gaming K70 LUX Keyboard  as /devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2:1.1/0003:1B1C:1B36.0003/input/input21
Oct 30 00:14:42 desktop kernel: hid-generic 0003:1B1C:1B36.0003: input,hiddev0,hidraw2: USB HID v1.11 Keyboard [Corsair Corsair Gaming K70 LUX Keyboard ] on usb-0000:00:14.0-2/input1
Oct 30 00:14:42 desktop kernel: usbhid 3-2:1.2: couldn't find an input interrupt endpoint

```

Causing some erratic issues with the keyboard on boot sometimes.  Sometimes keyboard will be 100% unresponsive.  Rebooting will sometimes fix the issue.  Does not occur all the time, intermittent.

I have also had to modify the following file:
src/usb.h
```
#define P_K70_LUX       0x1b36
#define P_K70_LUX_STR   "1b36"
```

Output of lsusb
`Bus 003 Device 004: ID 1b1c:1b36 Corsair `

Keep up the great work, and let me know if any more info is needed.

Some basic info:
```
[root@desktop ckb-daemon]# uname -a
Linux desktop 4.8.4-1-ARCH #1 SMP PREEMPT Sat Oct 22 18:26:57 CEST 2016 x86_64 GNU/Linux
[root@desktop ckb-daemon]# dmesg
[    0.000000] microcode: microcode updated early to revision 0x20, date = 2016-03-16
[    0.000000] Linux version 4.8.4-1-ARCH (builduser@tobias) (gcc version 6.2.1 20160830 (GCC) ) #1 SMP PREEMPT Sat Oct 22 18:26:57 CEST 2016
[    0.000000] Command line: initrd=\intel-ucode.img initrd=\initramfs-linux.img root=PARTUUID=09dc119a-4fd9-4039-b5c3-f9b584ad9dc2 rw quiet splash usbhid.quirks=0x1B1C:0x1B36:0x20000408
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'standard' format.
[    0.000000] x86/fpu: Using 'eager' FPU context switches.
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x0000000000057fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000058000-0x0000000000058fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000059000-0x000000000009dfff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009e000-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x00000000c9dfdfff] usable
[    0.000000] BIOS-e820: [mem 0x00000000c9dfe000-0x00000000c9e04fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000c9e05000-0x00000000caa99fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000caa9a000-0x00000000caf5efff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000caf5f000-0x00000000de42cfff] usable
[    0.000000] BIOS-e820: [mem 0x00000000de42d000-0x00000000de4b9fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000de4ba000-0x00000000de500fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000de501000-0x00000000de63bfff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000de63c000-0x00000000deffefff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000defff000-0x00000000deffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000f0000000-0x00000000f7ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed00000-0x00000000fed03fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed1c000-0x00000000fed1ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000041dffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] e820: update [mem 0xc8f0f018-0xc8f2f457] usable ==> usable
[    0.000000] extended physical RAM map:
[    0.000000] reserve setup_data: [mem 0x0000000000000000-0x0000000000057fff] usable
[    0.000000] reserve setup_data: [mem 0x0000000000058000-0x0000000000058fff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000000059000-0x000000000009dfff] usable
[    0.000000] reserve setup_data: [mem 0x000000000009e000-0x000000000009ffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000000100000-0x00000000c8f0f017] usable
[... 797 lines cut ...]
[    8.233379] hid-generic 0003:0D8C:0005.0004: input,hidraw3: USB HID v1.00 Device [BLUE MICROPHONE Blue Snowball] on usb-0000:00:14.0-3/input2
[    8.345702] usb 3-4: new high-speed USB device number 6 using xhci_hcd
[    8.478528] hub 3-4:1.0: USB hub found
[    8.479466] hub 3-4:1.0: 4 ports detected
[    8.655745] usb 3-6: new full-speed USB device number 7 using xhci_hcd
[    8.955753] usb 3-11: new full-speed USB device number 8 using xhci_hcd
[    9.088551] input: Logitech Gaming Mouse G502 as /devices/pci0000:00/0000:00:14.0/usb3/3-11/3-11:1.0/0003:046D:C07D.0005/input/input23
[    9.088719] hid-generic 0003:046D:C07D.0005: input,hidraw4: USB HID v1.11 Mouse [Logitech Gaming Mouse G502] on usb-0000:00:14.0-11/input0
[    9.090770] input: Logitech Gaming Mouse G502 as /devices/pci0000:00/0000:00:14.0/usb3/3-11/3-11:1.1/0003:046D:C07D.0006/input/input24
[    9.146078] hid-generic 0003:046D:C07D.0006: input,hiddev0,hidraw5: USB HID v1.11 Keyboard [Logitech Gaming Mouse G502] on usb-0000:00:14.0-11/input1
[    9.259120] usb 3-12: new low-speed USB device number 9 using xhci_hcd
[    9.446637] hid-generic 0003:0764:0501.0007: hiddev0,hidraw6: USB HID v1.10 Device [CPS  CP 1350C] on usb-0000:00:14.0-12/input0
[    9.464320] mousedev: PS/2 mouse device common for all mice
[    9.477334] usbcore: registered new interface driver snd-usb-audio
[    9.696398] input: ckb1: Corsair Gaming K70 LUX Keyboard as /devices/virtual/input/input25
[    9.696914] input: ckb1: Corsair Gaming K70 LUX Keyboard as /devices/virtual/input/input26
[   12.456090] nvidia-modeset: Allocated GPU:0 (GPU-f4204ada-a424-b742-8e7a-f5a1ca619793) @ PCI:0000:01:00.0
[   12.636013] snd_hda_codec_hdmi hdaudioC1D0: HDMI: invalid ELD data byte 2
[   12.869382] snd_hda_codec_hdmi hdaudioC1D0: HDMI: invalid ELD data byte 1
[   13.162263] fuse init (API version 7.25)
[   17.155411] random: crng init done
[   21.410081] snd_hda_codec_hdmi hdaudioC1D0: HDMI: invalid ELD data byte 1
[   21.650099] snd_hda_codec_hdmi hdaudioC1D0: HDMI: invalid ELD data byte 16
[   79.069514] snd_hda_intel 0000:00:1b.0: IRQ timing workaround is activated for card #0. Suggest a bigger bdl_pos_adj.
```

jonathan-bird, 2016-10-31:
I've got mine plugged into the monitor & turning the monitor off n on fixes it.

nkkromhof, 2016-11-02:
My K70 Lux seems to be working reliably so far, except for when I press the brightness button... This will make all LEDs turn off and the keyboard becomes unresponsive until I unplug it from the USB port and plug it back in, at which points the lights are back on at their normal level as well.

There is no output in syslog when this happens.

Not sure if this is somehow related or not.

0mni, 2016-11-05:
If I reboot several times, I have found that my keyboard becomes completely unresponsive.  Requires a complete shutdown, killing power to the PSU.  Then the keyboard will respond again, until I reboot a few times.  Something is going on, but I don't know exactly what.