ccMSC/ckb #522: Scimitar bug

Cryphunter, 2016-11-13:
This is on the MAC.
While the software is active, my Scimitar bugs out on World of Warcraft. When i click to move my camera, instead of staying in the place i clicked the mouse moves in the direction I'm looking. but when i disable the software and replug my scimitar it doesnt bug out anymore. which wouldnt be a problem but i lose access to my side buttons. A thing to note is i not only need to quit CKB but also unplug and replug my scimitar. Please help if possible, might have to sell the mouse and get a razer or logitech if issue can't be fixed

Cryphunter, 2016-11-14:
If there is logs needed I can post them but I'm not the best with files and computers or else id try to fix it myself. so if there is some logs i need to post i am good at following directions

dylanvorster, 2016-11-15:
This might be related to #521 as I'm having a similar type of issue with my M65

Cryphunter, 2016-11-16:
Its possible. have you tried disabling software and then trying to do what you were trying to accomplish or do you need the software in order to complete it. I can possibly post a video of how my mouse acts while trying to play world of warcraft