ccMSC/ckb #524: URGENT! Corsair K70 Lux RGB

MrQuikenHD, 2016-11-15:
I heard there is a branch wich enables Corsair Rapidfire K70 (I suppose the lux will work aswell) 
The thing is I dont know how to use it :( If anyone could explain me it would be very helpful since I will have to give the keyboard back to the store if I can't get it to work on mac.
Also if you have already made a .app of that new-dev branch upload it to mediafire and give me a link please :) It would be easier. Thanks/

petrklus, 2016-11-16:
@MrQuikenHD just a note that this is not an urgent issue, it only references future functionality.

MrQuikenHD, 2016-11-16:
@petrklus I said urgent coz I might have to deliver the keyboard back to the store if I can't get it to work on OS X and since I don't have many days left till the deliver max date I needed a quick answer :/

mattanger, 2016-11-16:
I think its safe to say this getting fixed before your return date is
highly unlikely. No one appears to be actively supporting this project.

petrklus, 2016-11-16:
What I am trying to say @MrQuikenHD is that while this matter may be urgent to you, it is not urgent in a sense of this project. Urgent issue would be if a build of the app was causing kernel panics. 

And I second @mattanger - you are unlikely to get a new build in a short amount of time.

xsappyx, 2016-11-18:
I have a k70 lux working using the newdev branch of this repo. It takes about 20 seconds on first boot for ckb and udev to settle down and then everything works fine. I've tested with recent 4.7.10, 4.8.6 and 4.8.7 kernels the last couple weeks and all of them work fine. I'm not using any usb quirks kernel options.

MrQuikenHD, 2016-11-19:
@xsappyx DUDE! Please tell me how to use the newdev branch :( That's what I really need. And ASAP becoz my delivery date ends in 1 week. If you have Twitter follow me @Quiken_GD So we can Dm :)

MrQuikenHD, 2016-11-19:
@petrklus Sorry then :) I edited the title tho

V3XATI0N, 2016-11-27:
[ckb-newdev.zip](https://github.com/ccMSC/ckb/files/614965/ckb-newdev.zip)
This file contains a snapshot of newdev that works with K70 Lux. Not sure why but a build I did from the current state doesn't work.

syropian, 2016-12-02:
@V3XATI0N Do you really need to install 13gb worth of stuff for q5 to work? Seems ridiculous.