ccMSC/ckb #526: Collaboration on newer CKB

mattanger, 2016-11-20:
Hello to anyone interested, I'm, slowly, working on a CKB fork that hopefully will have some improvements. I think this is an awesome project, but as everyone has noticed.. it's maintainer @ccMSC doesn't seem to be able to support it. 

CKB in it's current state is basically stable for me, K70 RGB on Arch Linux. But, I've noticed some UI issues that I've been wanting to fix, and that's what I'm somewhat focused on now. I'm also cleaning up the codebase a bit and converting it to a CMake project. I won't be able to dedicate much time to fixing anything for the next two months, but towards January 2017 I'll have more time to work on this project. 

If there are any other developers who want to collaborate on supporting and/or fixing this project up let me know!

mattanger, 2016-11-20:
Perhaps I should have noted that I only have limited access to OSX.

fleischie, 2016-11-20:
This seems like a good idea and I'm willing to help.

I also have a K70 RGB on Arch. Let me know if I can help somehow. 😊

guenniman, 2016-11-21:
I'm also willing to help.

I have an Corsair STRAFE RGB on Linux Mint.

mattanger, 2016-11-21:
Thanks for the replies! I'm not sure what exactly will be need off the bat, but all help is welcome!

stephenhouser, 2016-11-21:
There have been other forks you should look at that have been active more recently than the original branch. In https://github.com/ccMSC/ckb/pull/437 we were working on adding better macro handling.

fleischie, 2016-11-21:
@stephenhouser Would you suggest creating a fork, that merges all currently un-merged forks/branches and starting from there?

@mattanger What is your general outline how and what to address? How do you want to address the issues with the driver?

mattanger, 2016-11-21:
@fleischie  I don't have much of an outline yet. I haven't been able to dedicate enough time to really figure out any sort of concrete plan. I've been working on some UI improvements to make CKB more usable on a high dpi display. 

I'm open to all suggestions on where to go with it. As a starting point, I think getting all the un-merged forks and branches merged and going from there is a great idea.

webdev23, 2016-11-28:
Hi everyone. I am currrently making a controller for the ckb-daemon. 
I am using electron with html javascript that talk to a php file served in localhost.
The web audio api is used to get system audio datas.
Here is a preview: https://www.youtube.com/watch?v=H7mKN2PASGM
The interface is currently fully remade and some more effects.
Preparing instructions and codes!
See a basic php/html setup, a rainbow running in daemon: https://github.com/webdev23/phpCorsair

guenniman, 2016-12-02:
I got my hands on a Scimitar Mouse for testing

tatokis, 2016-12-04:
K70 RGB on Ubuntu here. No Mac computers, but planning to set up a VM if possible and test in there.
Willing to help whenever I have time.

It would be appreciated if anyone could give some insight on what Apple changed on Sierra to break CKB.

Also, we should have a single fork, deprecating all of the existing ones (the WIP ones should become branches on the new fork), possibly rename the repo to something along the lines of "ckb-next", or "ckb-revived" so that people won't get confused.

@mattanger if you don't mind, give write access to the people interested in here on your fork, and we can start merging existing PRs that have been waiting for a long time, and then go from there.

fleischie, 2016-12-04:
I suspect the new TouchBar might have something to do with how the Keyboard System is structured in the newer MacOS Version. Although of course I am no expert and have no previous experience writing device drivers in MacOS.

I will take a look tomorrow, whether I find something insightful but I doubt this will be perfectly easy to find out.

mattanger, 2016-12-05:
@tatokis That sounds good to me. I'd planned on starting merging the PRs first and doing what you suggested. Soon as I get time, I'll grant some access to start merging in some PRs. For the next week I'm busy as hell, but, if you're looking to get started immediately, would you like to take lead on getting those PRs merged? Or anyone else who wants to get started lemme know, and I'll set up access so those PRs can get merged into my fork. Otherwise, I'll get started on it sometime next week. 

Also, it would be real nice if we could find a developer running OSX. I know little about how OSX handles keyboards, or what kind of funny stuff it does. 

@webdev23 that sounds like a neat project. One thing I would like to add is an easy way to define effects and create effects without running a binary.

mattanger, 2016-12-05:
@fleischie Good luck on figuring out the MacOS subtleties. I suppose keep us updated on anything you figure out, and *maybe* we'll be able to do something about Sierra...

fleischie, 2016-12-05:
@mattanger thankfully @ccMSC documented the `OS_MAC` portions of the device driver daemon, so that I can get a somewhat clear understanding, of the quirks inside the source code.

On the other hand most of the workarounds are due to the very non-standard behaviour of many things (e.g. mouse-acceleration, keyboard-layout, time-management, etc.)

I will try to make time to research some more of the Apple-specific documentation regarding device-drivers etc. but also to figuring out, why this project writes a device-driver at all (I suppose it may be because of more previous versions of Mac OS). Regarding the issues it seems the devices work more well-rounded without `ckb`, the lighting and macro-ing is implemented in userspace (as far as I can tell but may be wrong). BUT maybe I am missing the bigger picture, yet.

As I said before, I will try to make time as well to research more into this and try to understand the ifs-buts-and-whys on the sourcecode.

:muscle:

tatokis, 2016-12-05:
I am free to get work done this week, I'd just like to ask you to get your fork up to date/even with the main repo, as that's the only thing I am not sure how to do. 

Also, regarding Sierra, I've been looking at how Karabiner is dealing with it. 
Turns out, the dev started working on Karabiner Elements, which is a rewrite for Sierra, so that probably means quite a few things need to be changed in ckb. 

Maybe there is an official doc that explains what needs to be changed, but I doubt it. 

So, as a short term roadmap: 

-> merge existing PRs 
-> get in contact with people who have forks in development, and see if they are interested in helping out. 
-> figure out what exactly is broken on Mac and find a solution 
-> make a list of new devices that will have to be supported in the future, and see which ones we can get away with by just adding their USB ids. 
-> go through existing bugs and filter out the ones that along the lines of "plz help". 

Feel free to suggest more, to add them to the new readme as a reference point.

mattanger, 2016-12-05:
I'll see what I can add to the list when I get some time to think about it.
I think my fork is up to date, I know it is locally, but I don't remember
if I pushed any thing up or not. I should be able to check on it later
today. If its not I'll get it synced with the main repo.

I'll also put up a readme/doc thing for any new changes.

mattanger, 2016-12-05:
Two more things. One, I have not been keeping track of this, but does the keyboard work on Sierra without the ckb? Just with basic functionality? 

Also, I'm planning on getting another keyboard, suggestions on which one to get? One that may needs some support or features. Its for a windows PC so I don't care if it works with MacOS or *nix.

mattanger, 2016-12-05:
@fleischie @tatokis My fork is now up to date. I've also added a road map doc, https://github.com/mattanger/ckb/commit/9bd99dcfed75ed605019e83f4fbf1f9816117985. Just has @tatokis suggestions in it. I'll flesh it out some more when I have some time. Would both of you like write access?

tatokis, 2016-12-05:
Sure, that would be appreciated.

fleischie, 2016-12-06:
@mattanger I would appreciate the write-access, too. :+1: :smiley: 

I have to check whether the keyboards (the one, that I have) works on Mac OS Sierra (if I find time after work) without `ckb`. I doubt the devices work entirely different from standard-compliant keyboards of other vendors. Of course the extra-functionality might be unsupported (macro-keys probably, disable-windows-lock, etc.). But as I mentioned already, I have to check. :wink:

mattanger, 2016-12-06:
@fleischie @tatokis Just sent both of you collaborator invites. Added a thread for planning out the road map. Let's maybe move some of our discussions to one of those threads, and leave this open for anyone else interested? 

And @fleischie I can't imagine that they wouldn't be much different from any other compliant keyboard. I'll see what I can do about checking on that to. My fiance has a Mac.. but I'm not sure if it's running Sierra or not. If I can pry it off her for a little bit I'll see what testing I can do.

fleischie, 2016-12-06:
I have updated my gf's Mac to Sierra and checked the keyboard (older K70 RGB). The main keys seem to work without a problem (a-z, 0-9, num keys, modifiers, etc.), even the media keys (except stop) and the win-lock-key work.

So maybe this is something to consider in the future. 😄

Also thanks for the collaboration invitation. 🙏

grounded042, 2016-12-07:
I'm willing to help out, though I'm pretty rusty on C++/C. Would it be worth considering writing the base driver in C/C++ and then using a higher level language to communicate? I haven't looked in a while, but it used to be I could just write to a file to send commands to the keyboard for lighting which made it really easy to use a higher level language.

I could also help test. I have a K70 and K95 and use macOS

light2yellow, 2016-12-09:
@mattanger the thing is people will still be googling for *this* repo and there's no way to find your fork if one doesn't read issues. If you are going to do something serious I guess you should (if the licence permits) just start a new repo named ckb2 or whatever copying the needed parts of the code from here and credit the original author. So there's no reason to create a playground for 10 developers and 10 users (and 0 non-devs). Otherwise the fork will never reach masses.

tatokis, 2016-12-09:
That's why I suggested to just rename it. 

It can be renamed to something like ckb-revived and still be a fork, preserving the original commits and authors. 

When it gets to a stable state again, it can be posted on the corsair forums, and it will gain popularity that way. 

But for now, we should just work towards fixing some major issues, such as the bugs related to Sierra

mattanger, 2016-12-11:
I'm all about doing a name change! Though, I think we should have a few
improvements that distinguishes our fork.

mackermans, 2016-12-13:
@mattanger @tatokis 
Small step towards Sierra support: https://github.com/ccMSC/ckb/pull/541

How is that revival project coming along?

tatokis, 2016-12-13:
@mackermans Thanks, merged on the fork.
So far I've just been merging existing PRs and making small changes. Slowly getting there.

mattanger, 2016-12-13:
@tatokis seems to be making fantastic progress. Come tomorrow I'll be
contribute a lot more.

webdev23, 2016-12-18:
Still working on a controller for the ckb-daemon built with web technologies, here is a easy deploy command line: 
Audio equalizer, matrix effect, rainbow effect, can be chained. Works in background. Web pages can gives events to the keyboards.
> git clone https://github.com/electron/electron-api-demos && mv electron-api-demos phpCorsair! && cd phpCorsair! && git clone https://github.com/webdev23/phpCorsair.git && mv phpCorsair/* . && npm install && npm start

Current screenshot:
[image]

If you like, please give a star!

matiass, 2017-01-17:
I am also willing to help! 

Let me know!

frickler24, 2017-01-18:
Hi @matiass, tried to install it on a linux mint 18 KDE. When running, I get the following exception:
`~/Projekte/phpCorsair! $ **npm start**

> phpCorsair!@1.2.0 start /home/lutz/Projekte/phpCorsair!
> electron .

internal/child_process.js:298
    throw errnoException(err, 'spawn');
    ^

Error: spawn EACCES
    at exports._errnoException (util.js:870:11)
    at ChildProcess.spawn (internal/child_process.js:298:11)
    at Object.exports.spawn (child_process.js:362:9)
    at Object.<anonymous> (/home/lutz/Projekte/phpCorsair!/node_modules/electron/cli.js:7:18)
    at Module._compile (module.js:410:26)
    at Object.Module._extensions..js (module.js:417:10)
    at Module.load (module.js:344:32)
    at Function.Module._load (module.js:301:12)
    at Function.Module.runMain (module.js:442:10)
    at startup (node.js:136:18)

npm ERR! Linux 4.4.0-59-generic
npm ERR! argv "/usr/bin/nodejs" "/usr/bin/npm" "start"
npm ERR! node v4.2.6
npm ERR! npm  v3.5.2
npm ERR! code ELIFECYCLE
npm ERR! phpCorsair!@1.2.0 start: `electron .`
npm ERR! Exit status 1
`
I do not use node.js elsewhere.

webdev23, 2017-01-18:
@frickler24 Thanks for your feedback. I am the author of phpCorsair. I didn't test much on many systems, but ubuntu. It made it in one shot, and still using it every day.  Your feedback is precious for me. It look that you don't have npm installed. Try **sudo apt install npm** 
Please for support and requests, let's continue on the phpCorsair repo, tx!  https://github.com/webdev23/phpCorsair