ccMSC/ckb #527: Working Binary Package for k65 LUX on OS X

H-Reed, 2016-11-22:
Got a k65 today and can't seem to get it up and running to it's fullest. I see the new dev branch and would like a binary package if its available. The latest version is still not connecting to my keyboard. Thanks!

luismaqos, 2016-11-22:
SO DID U GET it to work cuz mms says no device detected

H-Reed, 2016-11-22:
Not yet. I am going to recompile RKB later tonight and I'll update you. Would love a working Binary for the k6k Lux tho...

jolosantos, 2016-11-22:
Try using this recompiled ckb from the `newdev` build. You should follow [this megathread](https://github.com/ccMSC/ckb/issues/386)

[ckb-newdev-build.zip](https://github.com/ccMSC/ckb/files/576975/ckb-newdev-build.zip)

H-Reed, 2016-11-22:
I've tried that one several times, I keep getting re-directed to that thread and branch. It hasn't worked sadly. My plan (as of now) is to do a recompile (just using the build from source instructions) and do a soft reset of the keyboard itself. If not, I will find a friend with windows and have him save a preset straight to the keyboard, that way I can get something at least. Hope it works out. I still would love a set binary that would just work.....but that apparently doesn't exist as of now. So we'll have to see.

H-Reed, 2016-11-23:
None of the above worked. Still not detecting my keyboard :(

ags911, 2016-11-23:
I also tried that build and it did not launch at all. Interestingly enough your version has a ckb-daemon file inside the resources folder yet mine does not. I also cannot get mine to detect my M65 Pro. The newdev version I compiled opens fine and shows beta v0.2.6+d01 but gives "No devices connected". I'm guessing it should have the ckb-daemon file inside the app but I followed all instructions for it. Any ideas?

[CKB Zipped.zip](https://github.com/ccMSC/ckb/files/609129/CKB.Zipped.zip)

H-Reed, 2016-11-28:
It still does't work for me. Im pretty upset, as I have tried everything. I guess I will wait for a binary or a new update.

CarlaHeywood, 2017-01-31:
When I try to build the project in QT two redefinition errors occur.
CLOCK_MONOTONIC from os.h and time.h
clockid_t with different typesfrom os.h and time.h. ('int' vs 'enum clockid_t')

tatokis, 2017-01-31:
There's a fix in https://github.com/mattanger/ckb-next for that issue.

CarlaHeywood, 2017-02-05:
I tried this install as well. I tried to build it and I also tried to install from the package. My keyboard is not detected. I have the k65 lux RGB. I was using CKBnewDev before and it worked fine. I had to reinstall the operating system on my iMac so I am just trying to reinstall this software to use my keyboard again. I don’t know whats going on now. When I try to build it in QT it is giving the same errors as before.

CarlaHeywood, 2017-02-05:
Also, when I try to follow the read me, going to the ckb.pkg in the downloads, it is actually not there. Under v0.2.4. I only see Source code(zip) and (tar.gz)

frickler24, 2017-02-05:
@CarlaHeywood Do you still look at ccMSC/ckb?
Please have a look at [the new fork by mattanger](https://github.com/mattanger/ckb-next). 
It is called ckb-next, current version is v0.2.7 and the current work is done there, because for ccMSC/ckb we do not have a maintainer anymore.

CarlaHeywood, 2017-02-05:
It works!! Thanks so much!! I tried to build it in QT first. It created the .app. It was not recognizing it still but then I used the quick install and that worked! 

Thanks for you time!