ccMSC/ckb #528: device not recognized RGB Mouse / OSX ckb beta v0.2.6

rewbrennon, 2016-11-22:
There are no options for the mouse, and ckb says device not recognized
ckb beta v0.2.6
osx 10.12.1

rewbrennon, 2016-11-22:
[image]

Featherw8, 2016-11-25:
Man did you find a solution for this i am facing the same problem!

Neverland, 2016-11-28:
You know that's true, and so do I

startupfoundry, 2016-12-19:
I'm having the same issue.

iamjustgodly, 2016-12-24:
same for me as well but i use k70 rgb lux key board

DragonLegends, 2017-01-09:
Okay, so this program runs of the ckb-daemon. This program automatically launches at the startup of the computer, and allows ckb to detect the mice/keyboards. Unfortunately, on the shutdown or restart of the computer, it mess with (As far as I know) the mices' sensitivity and other things. Sometimes, it will drag the mouse to the top left of the screen if you click and drag. Other times it will not run properly with certain programs. The only way to stop these errors, is to use activity monitor to quit the program. The only problem is that it will not relaunch till the computer restarts. I don't know how to fix any of this, but this is what is causing your problems.

Message the developer about the ckb-daemon's problems. If you find a fix, please let me know.