ccMSC/ckb #529: New device ID for K70 Lux 0x1b36

hevanaa, 2016-11-26:
A new K70 LUX blue (not RGB) has a different device ID than what a keyboard with the same name has in the newdev branch. The newdev branch has "P_K70_LUX       0x1b33". This new keyboard has otherwise identical name etc, but the device id is 0x1b36. By default, ckb (compiled from the newdev branch) wasn't able to detect the keyboard. When changing (or adding) the ID to 0x1b36 in the correct places, ckb-daemon was able to identify the device and when manually starting the daemon, also the ckb GUI utility worked. Also the multimedia keys work at this point. The big issue is however that when booting, at the point where ckb-daemon is loading, the keyboard always stops working. The only way to get the daemon to work is to (including resetting the keyboard and booting from a powered off PC) boot in BIOS mode, login and then manually start the daemon.

Some info:

Operating system:
Fedora 24
Kernel 4.8.8

lsusb|grep Corsair
Bus 003 Device 005: ID 1b1c:1b36 Corsair

relevant info from dmesg:
[    1.493216] usb 3-6: new full-speed USB device number 3 using xhci_hcd
[    7.149040] usb 3-6: unable to read config index 0 descriptor/all
[    7.149100] usb 3-6: can't read configurations, error -110
[    7.304934] usb 3-6: new full-speed USB device number 5 using xhci_hcd
[    7.474800] usb 3-6: New USB device found, idVendor=1b1c, idProduct=1b36
[    7.474805] usb 3-6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    7.474807] usb 3-6: Product: Corsair Gaming K70 LUX Keyboard 
[    7.474810] usb 3-6: Manufacturer: Corsair
[    7.474812] usb 3-6: SerialNumber: 16030015AEB89C4755BB595DF5001940
[    7.476806] input: Corsair Corsair Gaming K70 LUX Keyboard  as /devices/pci0000:00/0000:00:14.0/usb3/3-6/3-6:1.0/0003:1B1C:1B36.0001/input/input6
[    7.477138] hid-generic 0003:1B1C:1B36.0001: input,hidraw0: USB HID v1.11 Keyboard [Corsair Corsair Gaming K70 LUX Keyboard ] on usb-0000:00:14.0-6/input0

Adding usbhid.quirks=0x1B1C:0x1B36:0x0x20000408 doesn't help, keyboard only seems to stop working after the kernel loads, can't even use it in single mode.

For now the keyboard can be used in BIOS mode only. Of course multimedia keys don't work.

Could it be a problem with the USB protocol, or could this be easily patched somehow? I'll still try to debug the bootup problem.

hevanaa, 2016-11-26:
OK, well then, I see when searching a little more that indeed 0x1b36 has been discussed. I'll try the workaround.

hevanaa, 2016-11-26:
Doing the following change in usb.h didn't help:

#define P_K70_RFIRE     0x1b36
#define P_K70_RFIRE_STR "1b36"

It was basically the same thing I had tried already.

hevanaa, 2016-11-26:
Another note. The Windows key doesn't work with the driver running.

hevanaa, 2016-11-26:
Of course, usbhid.quirks doesn't work with this device ID, because there is no support for it in the kernel. I tried adding to the kernel source and rebuilt hid modules:

drivers/hid/hid-ids.h:
#define USB_VENDOR_ID_CORSAIR           0x1b1c
#define USB_DEVICE_ID_CORSAIR_K70R      0x1b09
**#define USB_DEVICE_ID_CORSAIR_K70LUX    0x1b36**
#define USB_DEVICE_ID_CORSAIR_K95RGB    0x1b11

drivers/hid/usbhid/hid-quirks.c:
        { USB_VENDOR_ID_CORSAIR, USB_DEVICE_ID_CORSAIR_K70R, HID_QUIRK_NO_INIT_REPORTS },
        **{ USB_VENDOR_ID_CORSAIR, USB_DEVICE_ID_CORSAIR_K70LUX, HID_QUIRK_NO_INIT_REPORTS },**
        { USB_VENDOR_ID_CORSAIR, USB_DEVICE_ID_CORSAIR_M65RGB, HID_QUIRK_NO_INIT_REPORTS },
        { USB_VENDOR_ID_CORSAIR, USB_DEVICE_ID_CORSAIR_K95RGB, HID_QUIRK_NO_INIT_REPORTS | HID_QUIRK_ALWAYS_POLL },

But I didn't notice any difference with:
usbhid.quirks=0x1B1C:0x1B36:0x20000408

hevanaa, 2016-11-26:
Added initial support for this keyboard #530

hevanaa, 2016-11-29:
Updated keyboard firmware to 2.0.4 when I finally got a Windows PC connected to it. Now most things seem to work and the driver also works mostly. Scan mode can now be used. The right Windows button still doesn't work at all and its toggle button doesn't do anything. The ckb settings program freezes when clicking on certain things, e.g. key colors. I guess the keyboard is identified as an RGB keyboard, even though it is single color.

Spencer-H, 2016-12-01:
Does your patch function with or without the kernel patch you posted here?

hevanaa, 2016-12-01:
Works fine without the kernel patch. That was just me messing around when the keyboard didn't work. Most of the problems were due to the 1.x firmware. The 2.0.4 firmware is really required.

tatokis, 2016-12-01:
The patch just does the same thing the quirks do.
The point of usbhid.quirks is so that you can tell the kernel what 
quirks to apply to what device without having to patch and recompile it.

Spencer-H, 2016-12-01:
Alright. I just bought the red variant of that and it has the same ID, it
works, sort of, with the default driver (ckb doesn't recognize it), but I'm
going to update the firmware and try it with yours.

hevanaa, 2016-12-02:
Not exactly relevant for this issue, but I've created rpm packages for Fedora and Epel of ckb at [https://copr.fedorainfracloud.org/coprs/johanh/ckb/](https://copr.fedorainfracloud.org/coprs/johanh/ckb/)

Spencer-H, 2016-12-02:
> org/coprs/johanh/ckb/

The keyboard works perfectly after the firmware update and with the patched
driver.

midnightconman, 2016-12-09:
Looks like there are multiple product id's for the k70 lux:

I have `1b33`, not `1b36`

```
[  218.504150] usb 2-1.4: USB disconnect, device number 4
[  221.776389] usb 2-1.4: new full-speed USB device number 6 using ehci-pci
[  221.871452] usb 2-1.4: New USB device found, idVendor=1b1c, idProduct=1b33
[  221.871457] usb 2-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  221.871460] usb 2-1.4: Product: Corsair Gaming K70 LUX RGB Keyboard
[  221.871463] usb 2-1.4: Manufacturer: Corsair
```