ccMSC/ckb #531: Hardware not detected on mac

jhcom, 2016-11-29:
Hello, I really followed all the steps. I have the latest firmware and latest version from git, but somehow the app can't recognize my k70 keyboard and m65 mouse. I have installed Qt and xcode is working. 

I am really not sure what I am doing wrong. Could you help me out?

MrQuikenHD, 2016-12-01:
First of all make sure your QT has the latest version. Second, after you have qt and xcode installed you most compile this "new-dev branch" github.com/ccMSC/ckb/tree/newdev
Download the zip and then extract it to the desktop. After that you must run the "quickinstall"
It will open terminal and you will have to wait while it compiles. In the end it will ask you if you want to start it now etc... Click Y 3 times. After that you may want to reboot. After rebooting you will find a new ckb app on the "new-dev" folder. Open that and ta-daa :)

jhcom, 2016-12-16:
Hello,

Thank you for the reply, but I really can’t seem to get it working. I receive this error and then a key is asked. I attached an image of it. I tried writing 1x Y, 3x Y and my password, nothing seems to work. Could you please tell me what I am doing wrong?

I use the following hardware.

Corsair Gaming K70 RGB RAPIDFIRE Keyboard :

  Product ID: 0x1b38

Corsair Gaming M65 Pro RGB Mouse:

  Product ID: 0x1b2e

I hope to hear from you soon!

Kind regards,

Julian Hana

[cid:2DADC4E3-D405-45DA-9A64-FB5EA9BA16A0]