ccMSC/ckb #533: K70 Lux RGB Not Detected

syropian, 2016-12-01:
macOS Sierra 10.12.1

Tried:

- Using different USB Port
- Uninstalling and reinstalling the driver
- Restarting app
- Restarting computer

Using ckb beta-v0.2.6 downloaded and installed through the pkg.

syropian, 2016-12-01:
Hm, restarted, then deleted and reinstalled the driver and now it's working 😅

hevanaa, 2016-12-02:
Update to latest 2.x firmware using a Windows PC. If you are on 1.x there might be issues.

syropian, 2016-12-02:
I don't have access to a Windows machine. However, I compiled the newdev branch and I _think_ it's working for now, though I didn't do any extensive testing. When I get home and replug my computer in I'll check if it's working. If so, I'll close the issue.

nylki, 2016-12-18:
@syropian Yep, Using the `newdev` branch works, just tried it!

closed 2016-12-18 by syropian