ccMSC/ckb #534: Shift Select or Cmd Select not Working on macOS Seirra

hessercan, 2016-12-01:
When ever Selecting multiple things on macOS Seirra, the normal hold shift or cmd doesn't work anymore, only an issue on new OS

lhqiang, 2016-12-02:
The same problem，macOS Sierra,  keyboards: Strafe

Ranhiru, 2016-12-02:
Same issue with me. macOS Sierra. Keyboard is K95 RGB. 

The funny thing is that Cmd key is working for Cmd+Tab to switch between apps. However it doesn't work for selecting multiple files or Cmd+Click in Chrome to open in a new tab.

zyxep, 2016-12-05:
I can confirm these issues with my K95 RGB.

I get a memory leak sometimes when i hit caps lock, i need to reboot to get it back to normal.

mhennessie, 2016-12-07:
Use the keyboard without CKB. This is an issue with all keyboards and Sierra. It was reported quite a while ago.